Akaflieg-Freiburg/enroute #337: For safesky traffic direction is not shown

sugacube, 2023-07-27:
When connecting to safesky to display traffic, aircraft are displayed as a green circle.

The enroute user manual however shows other aircraft as a green arrow indicating their direction of flight.
https://akaflieg-freiburg.github.io/enrouteText/manual/02-steps/traffic.html

**To Reproduce**
Steps to reproduce the behavior:
1. Install and start safesky app, enable traffic sharing to enroute
2. make sure to select maximum of traffic sharing
3. click take off button in safe sky
4. Start enroute
5. View other aircraft -> displayed a green circle in stead of green arrow

**Expected behavior**
Expects not a green circle but a green arrow as in the enroute manual

**Screenshots**

**Smartphone (please complete the following information):**
 - Device: medion lifetab (tablet) , moto g 5g (smartphone)
 - OS: Android 11
 - Enroute : v2.28.2

kebekus, 2023-07-27:
@sugacube I am aware of the problem. Wondering why Enroute Flight Navigation does not show the projected flight route for traffic reported by SafeSky, I analysed the GDL90 output of SafeSky. I believe that I found an issue: SafeSky produces non-compliant GDL90 output.

The problem is with the 4-bit field "m = Miscellaneous Indicators" of ownship/traffic messages, as specified in §3.5.1.5 of the GDL90 specification (I attach a copy).

-    For ownship reports, the Bits 1+0 are set to %01, indicating that the field "tt" contains the ownship's true track. This is the correct behaviour, I believe.
-   For traffic report, the Bits 1+0 are set to %11, indicating that the field "tt" contains the ownship's true heading. I strongly doubt that true heading is really what SafeSky wishes to report.

Writing Enroute, we have tried hard to stick to specifications wherever possible. As a result, Enroute disregards the heading information, for which it has no use [wind correction angles can be substantial for slow aircraft, such as paragliders]. Working around the issue is difficult for us because Enroute typically cannot identify SafeSky as the source of the GDL90 data stream.

Best,

Stefan

kebekus, 2023-07-27:
@sugacube I informed the SafeSky developers on 06 Jul 23, but have not heard back from them.

sugacube, 2023-07-28:
@kebekus thanks for clarifying in detail and for the reference, gave me the opportunity to dive into the protocol. I can understand, imho you're very strict here, maybe rightfully so. i think other software may be less strict, maybe even ignores the m field.

Lets hope safesky looks into it.

kebekus, 2023-07-28:
I am closing this now because I do not know what else to do. Feel free to re-open once new information comes in.

closed 2023-07-28 by kebekus