Akaflieg-Freiburg/enroute #345: Vertical Airspace display

fbbln, 2023-09-25:
Especially in complex airspaces like around big airports or e.g. around Schleswig(DE), you often see vertically staggered airspaces.
It's hard to keep track of how high can you actually go without for example violating airspace C.

Therefore I suggest a small indicator showing the next airspace above, and maybe below one together with their respective height.

Suggestions:
[image]
or
[image]
or with current airspace also shown
[image]

This could be combined with #338 and #262

kebekus, 2023-09-27:
@fbbln Excellent idea! I put the feature on my TODO list. Note, however, that development is currently very slow because of a lack of manpower. I would like that feature, but it will take a long time to implement. Any help is incredibly welcome.

kebekus, 2023-10-10:
@fbbln I accidentally sent you a GitHub invitation that was meant for someone else. I canceled the invitation. Please do not take offense! Of course, if you'd be highly welcome if you are interested in joining the development team.

kebekus, 2025-09-21:
A full side view of the airspace situation is now available in version 3.0.0, which appeared on Google Play a few minutes ago.

@fbbln Thanks again for all your help!

closed 2025-09-21 by kebekus