Akaflieg-Freiburg/enroute #347: Error creating Geotiff files

a533, 2023-10-03:
**Describe the bug**
Unable to create Geotiff files

**To Reproduce**
Steps to reproduce the behavior:
1. Install GDAL, last version on Windows 10
2. type gdal_translate -of GTiff -a_srs EPSG:4326 -a_ullr -6,620555 43,982777 -5,736666 43,387222 input.png output.tif
3. gdal_translate finishes with no error and creates output.tif
4. Copy the file to the phone and open it with "Enroute flight simulator"
5. The app says "the file ...tif seems to contain an image without georeferencing information."

**Expected behavior**
That the output.tif file would be imported

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Browser n/a

**Smartphone (please complete the following information):**
 - Device: Lenovo
 - OS: Android 12
 - Browser n/a

**Additional context**
I include the output.tif file

Suggestion
As the Geotiff file is 20 times bigger than the input jpg, It could be nice if Enroute Fligth simulator could import jpg with the coordiantes added as exif metadata using exiftool, for insntance

exiftool -GPSLatitude="42/1, 15/1, 30/1" -GPSLongitude="71/1, 4/1, 20/1" -GPSLatitudeRef="N" -GPSLongitudeRef="E" image.jpg

Thank your for the excellent app and regards,

Alvaro
[output.zip](https://github.com/Akaflieg-Freiburg/enroute/files/12793208/output.zip)

a533, 2023-10-03:
Hi again.

This file (see file attached) has created with gdal_translate .exe using this command line: 

gdal_translate -a_srs EPSG:4326 -a_ullr <upper_left_longitude> <upper_left_latitude> <lower_right_longitude> <lower_right_latitude> input.png outpu
[output.zip](https://github.com/Akaflieg-Freiburg/enroute/files/12794973/output.zip)
t.tif

if I import it in QGIS it shows perfectly in top of a OpenStreetMap layer at the right location

But in Enroute Flight Simulator it says ""the file ...tif seems to contain an image without georeferencing information.""

Regards,
Alvaro

kebekus, 2023-10-04:
@a533 Thank you for opening this request. I understand you have a valid use case, and I would like to support the GeoTIFF file you created. We will look into this and report any progress via this channel. However, we will need some time as volunteers run this project. We lack the manpower to implement all feature requests immediately.

kebekus, 2023-10-04:
@ bloech1 Dear Heinz, could you please have a look at the GeoTIFF file Alvaro has generated and see if we can extend the GeoTIFF parser to support this file? -- Thank you!

kebekus, 2023-10-04:
@a533 Do you understand the GeoTIFF format well? To be honest, we struggle to make sense of the specifications. We would be grateful for any help that we can get.

bloech1, 2023-10-04:
Hallo Stefan,

 

soeben korrigiert, auch wenn ich es eher als einen Fehler von GDAL ansehe. Sollte nun funktionieren.

 

Viele Grüße, Heinz

a533, 2023-10-04:
Thanks a lot, I'll wait for the next release,

Regards

AQ

kebekus, 2023-10-05:
@bloech1 Thank you for the extremely prompt reply. I will need about two weeks to pull the new code into the project and publish a new release. Until then, I'd like to thank you for your patience.

kebekus, 2023-10-05:
@a533 I would expect that many of our users would like to create GeoTIFFs like you did. I have no experience with the program, and I do not have access to a Windows machine. Would you be willing to write a short section for our user manual, perhaps including a few pictures?

a533, 2023-10-05:
> @a533 I would expect that many of our users would like to create GeoTIFFs like you did. I have no experience with the program, and I do not have access to a Windows machine. Would you be willing to write a short section for our user manual, perhaps including a few pictures?

Of course, I'll will

Regards
AQ

a533, 2023-10-05:
Good afternoon,

I'm attaching a georeferenced TIF file in compressed format. It only occupies 1.5 MB compared to the 12 MB of the uncompressed version. Mr. Heinz, can you please test if this compressed version works with the modifications you've made?

I've created a small program in Windows that allows you to:

1. Select a PDF file on your computer (VFR and IFR approach charts are downloaded in PDF format from each country's AIP).
2. If the PDF file has more than one page, a viewer allows you to navigate through the pages to select the one you want to use.
3. Convert the selected page to a PNG file and display it in a viewer.
4. Graphically crop the file to remove margins, legends, text, etc.
5. Mark 2 points on the map and enter the coordinates of those points. You can select a VOR, NDB, etc., with known coordinates or choose the intersection of parallels and meridians with exact numbers, e.g., N53°40'00" and W5°30'00".
6. Export it as a georeferenced TIF.

I have it almost finished, and I will share it shortly. If you're going to a new airport and want to georeference 4 or 5 charts, it's important that the whole process doesn't take more than 2 or 3 minutes.

However, the user needs to have GDAL installed on their computer since my program uses a command line "gdal_translate.exe" in the background for the conversion. GDAL occupies 170 MB, and the installation may not be straightforward for an average user. GDAL must be complete for it to work. I've tried leaving only "gdal_translate.exe" and a few libraries, but it stops working if the full program isn't present.

An alternative is to use "exiftool.exe." It's just one 8 MB file, easy to copy, install, etc.
[Leas_VOR.zip](https://github.com/Akaflieg-Freiburg/enroute/files/12819167/Leas_VOR.zip)

In the long run, it would be great if Enroute Flight Navigation allowed the use of both "TIF" files generated with GDAL and JPG files generated with exiftool. Exiftool allows saving only one set of coordinates, so you would need to use, for example, the TAG "Description" to include the two sets of coordinates. For example:

`ExifTool.exe -Description="West=-6.620555 North=43.982777 West=-5.736666 South=43.387222" input.jpg`

In the metadata of the JPG file, you would something like:

`...
Photoshop Format: Progressive
Progressive Scans: 3 Scans
XMP Toolkit: Image::ExifTool 12.67
About: uuid:7030863b-6285-11ee-ac70-dff9fe150b7d
Description: West=-6.620555 North=43.982777 West=-5.736666 South=43.387222
DCT Encode Version: 100
...
`

Best regards,

Alvaro

kebekus, 2023-10-07:
@bloech1 I have started to work on the GeoTIFF class in the repo geoImages. As a first step, I will make GeoTIFF compatible with a new API that I use for all classes that handle foreign file formats. Once this is working, I will simplify and robustify the GeoTIFF code.

If you have the time, it would be great if you could collect a few interesting TIFF files (big/small TIFF, different endianness, etc etc), so we can set up a few automated tests. I am of course worried that my "simplifications" might introduce new errors…

kebekus, 2023-10-07:
@bloech1 The header file "GeoTIFF.h" is now in final format. I will now turn to GeoTIFF.cpp and see how much I can simplify the code.

kebekus, 2023-10-16:
@a533 @bloech1 I have uploaded program version 2.29.2 to the Google Play/Beta. If you are subscribed to the beta, you can upgrade to that version now. The problem that@a533 describes should be fixed, and I am closing this issue now. If problems remain, please do not hesitate to re-open the issue!

Thank you for being so helpful!

@a533 If you want to write a paragraph for the manual ("How to create GeoTIFF files"), please text me by e-mail. I accept any format, though simple text or markdown would be most effortless.

@bloech1 Danke für die GeoTIFF-Sammlung. Ich komme in den nächsten Tagen auf dich zu.

closed 2023-10-16 by kebekus

a533, 2023-10-17:
@kebekus I sent you an email with instructions.
The beta works great. Thanks a lot.
Alvaro

kebekus, 2023-10-17:
@a533  Mail arrived, thanks. I will reach out to you in a few days.