Akaflieg-Freiburg/enroute #348: Error when connected to Stratux

VirusPilot, 2023-10-11:
**Describe the bug**
since the last iOS update (I believe) enroute is constantly messaging errors related to the traffic source (Stratux providing FLARM NMEA). This was first detected in SkyDemon but now also appears in enroute

**Smartphone/Tablet (please complete the following information):**
 - iPad or iPhone
 - iOS 17

**external device:**
Stratux version 1.6r1-eu29

If you need a Stratux device for testing, I can provide one.

VirusPilot, 2023-10-11:
SkyDemon reports "device connection lost". In enroute I am still trying to reproduce the error from yesterday but it was "Warnung, Problem mit dem Verkehrsdatenempfänger" - currently there is not much FLARM traffic here at night but my two FLARM sender devices (T-Beam and T-Echo) won't show up at least. Any chance that I could increase the area so that I could see ADS-B traffic?

VirusPilot, 2023-10-19:
Update: a Stratux has been shipped to SkyDemon - hopefully they can identify whether there is a problem with the latest versions of iOS or SkyDemon or both. So let's put my issue in the parking lot for the moment ...

schneisi, 2023-10-20:
Thank you for the update.
I currently don't have the time do dig into that myself.

kebekus, 2024-05-18:
Dear @VirusPilot  ,

are there any news on this? After scratching my head for a very long while, I came across a problem that may perhaps explain your problems with the Wi-Fi connection: Some devices do not send the NMEA sentences in one piece, but in several blocks (which do not even align with the boundaries of the sentences). I never anticipated this, and so Enroute was not able with this transmission mode. I have upgraded the app now.

Could you perhaps test the current beta version (=2.30.22), which is now available on Google Play?

Regards,

Stefan.

VirusPilot, 2024-05-18:
SkyDemon has resolved the issue and at this point enroute (iOS version) does not show an error any more. I can't test Android unfortunately. Not much traffic right now ...

kebekus, 2024-05-19:
@VirusPilot Thanks. I am closing the issue then. Feel free to re-open if the problem persists or comes back. 

Do not hesitate to send more reports in the future, whenever you find anything!

closed 2024-05-19 by kebekus

VirusPilot, 2024-05-19:
Sure, will do.