Akaflieg-Freiburg/enroute #351: MacOS Binary Artifacts from Github Actions?

VirusPilot, 2023-10-19:
I understand that there is no downloadable MacOS binary available but instead of building it individually I suggest that such binaries are made available as "artifacts" in the "Compile on Mac" action.

kebekus, 2023-10-20:
@VirusPilot I do not have access to an Apple computer, I do not know how to turn the compiled binaries into a DMG, and I won't be able to do any testing. Would you happen to know how to do these things? If so, could you suggest a suitable GitHub action? I will be more than glad to follow your suggestions.

kebekus, 2023-10-20:
@VirusPilot We are interested in offering Enroute for Apple desktop computers, but so far we lack the necessary manpower to maintain an Apple port. Would you be interested in joining the development team?

schneisi, 2023-10-20:
I currently don't have the time resources for it, but I will take a look as soon as I have the time

VirusPilot, 2023-10-20:
> @VirusPilot We are interested in offering Enroute for Apple desktop computers, but so far we lack the necessary manpower to maintain an Apple port. Would you be interested in joining the development team?

Well, I am not a real developer anymore so let me first try to create binaries from Github Actions. I have successfully created such binaries in my SoftRF fork so maybe I just just try here as well.

kebekus, 2023-10-20:
@VirusPilot Thanks. Looking forward to that!

VirusPilot, 2023-10-22:
So the first MacOS build artifacts can be downloaded in my fork but I still need to understand what is missing to build a MacOS app.

kebekus, 2023-10-23:
@VirusPilot Does this help: https://doc.qt.io/qt-6/macos-deployment.html

VirusPilot, 2023-10-24:
@kebekus Yes, will look into it ...

kebekus, 2023-11-20:
@VirusPilot Enroute is a fairly complex build, so I experimented first with Enroute's smaller brother, the "Add Times" app. Could you please look at the automatically-generated "developer builds" and let me know if these work for you?

- https://github.com/Akaflieg-Freiburg/addhoursandminutes/releases/tag/developerBuilds

VirusPilot, 2023-11-20:
Thanks for the hint, I will look into it. My daytime job is currently keeping me busy so this may take some time.

mipastgt, 2023-11-25:
I gave the above developer build a try on my Mac and I was able to install and run the program. However, I was not able to do anything reasonable with it, but this may be due to the fact that I neither know what this program is supposed to do nor what it is supposed to look like when it works properly.

I could offer some help in testing, as I am also interested in a Mac version of the program. I could also provide some help on building a fully customized DMG installer based on a working .app file. I have automated the icon, background and DMG creation with Retina support. If you are interested, just let me know. I have attached an image of my customized DMG installer for the upcoming new release of the AIP Browser DE in order to give you an idea how it could look like.

[image]

kebekus, 2023-11-25:
@mipastgt Thanks! The app is a little calculator app. Check the [website](https://akaflieg-freiburg.github.io/addhoursandminutes/), where you can also run the app inside your browser.

Thanks also for your offer to help with DMG creation. I am working on compiling Enroute for MacOS, but since we moved to new versions of the underlying libraries, this is not straightforward and will take a few weeks. I will keep you updated via this channel.

mipastgt, 2023-11-25:
Ok, so the functionality seems to work and looks identical to the output of the web version. But the Menu in the upper left doesn't seem to work every second time when you start the application. When it works, the corresponding texts are displayed and can be closed by pressing the OK button. But the next time you start the application it looks like this:
[image]
[image]
After another restart it works again and so on.

kebekus, 2023-11-26:
@mipastgt Thanks! I will come back to you, but this will take a while…

VirusPilot, 2023-12-27:
> @VirusPilot Enroute is a fairly complex build, so I experimented first with Enroute's smaller brother, the "Add Times" app. Could you please look at the automatically-generated "developer builds" [...]

No, even with some modifications I can't get gh actions to produce something reasonable. Giving up for the time being ...

kebekus, 2023-12-29:
@VirusPilot You can find an experimental binary for macOS her: https://v2202001110709105590.ultrasrv.de/nextcloud/index.php/s/sTzMTL7N9j67ose

Automatic generation via a GitHub action still doesn't work, though.

VirusPilot, 2023-12-29:
Won't launch on my iMac 14.2.1 (23C71): "Du kannst das Programm „enroute.app“ nicht öffnen, da dieses Programm auf diesem Mac nicht unterstützt wird."

mipastgt, 2023-12-29:
This version of the enroute.app is for ARM only. It's not universal like the previous one from Dec. 17th.

VirusPilot, 2023-12-29:
Ah so where can I find the Intel version from Dec. 17th?

kebekus, 2023-12-29:
I don't have that anymore. Will try to make a universal app. Give me a day or two.

VirusPilot, 2023-12-29:
Yes, awesome!

mipastgt, 2023-12-29:
Yes, here too.

VirusPilot, 2023-12-29:
I even connected my Stratux and can see traffic on the map.

kebekus, 2023-12-30:
Very well. I will work on the GitHub Actions, but currently, none of the macOS runners seem to work.

mipastgt, 2023-12-30:
Have you checked your spending limits? https://docs.github.com/en/billing/managing-billing-for-github-actions/managing-your-spending-limit-for-github-actions

kebekus, 2024-03-13:
Implemented. GitHub will automatically generate "Developer Builds" for a variety of platforms, including macOS. You find the binaries at https://github.com/Akaflieg-Freiburg/enroute/releases/tag/developerBuilds

closed 2024-03-13 by kebekus

VirusPilot, 2024-03-13:
Perfect, you made my day.