Akaflieg-Freiburg/enroute #353: Not receiving traffic heartbeat

papafox82, 2023-10-31:
I updated to the latest beta version on 28/10/23 prior to a flight that day. I was able to connect to the SkyEcho and receive GPS position, but it was unable to detect the traffic heartbeat. Another phone running the latest main release (v2.29.3) received traffic as normal however. Bringing the SkyEcho home to troubleshoot, Enroute (public release) was installed on another phone and confirmed to work as normal. I then uninstalled Enroute beta, and installed the public release on my phone again, and this has restored traffic reception, ruling out any connection issue with my physical phone and the SkyEcho. I suspect something has broken with the latest beta version.

All phones were Samsung, mine being a S9 running Android 10

kebekus, 2024-05-18:
Dear @papafox82 ,

after scratching my head for a very long while, I came across a problem that may perhaps explain your problems with the Wi-Fi connection: Some devices do not send the NMEA sentences in one piece, but in several blocks (which do not even align with the boundaries of the sentences). I never anticipated this, and so Enroute was not able with this transmission mode. I have upgraded the app now.

Could you perhaps test the current beta version (=2.30.22), which is now available on Google Play?

Regards,

Stefan.

kebekus, 2024-06-08:
Closing for lack of communication. Please feel free to re-open anytime.

closed 2024-06-08 by kebekus