Akaflieg-Freiburg/enroute #355: Expand flight vector to 6 mins

kebekus, 2023-11-01:
One user suggested the following:

> Der "Flugbahnvector" ist bei ihrem Programm auf 5 Minuten definiert. In meiner anderen schon erwähnten Flug - App 
habe ich diese "Heading - Line" auf 6 Minuten eingestellt! Der Grund ist recht banal. So sehe ich zum einen wo ich mich voraussichtlich in 6 Minuten befinden werde, zum anderen kann ich aber ohne viel zu rechnen auch die Distanz der Heading - Line entnehmen (6 min = 1/10 Std.; bei Gesch. = 110 kt. ist der Vorausstrich also 11 nm, bei 90 kt. = 9 nm usw.) Bei Positionsangaben via Funk macht es manchmal durchaus Sinn die Entfernung vom Punkt X anzugeben und nicht 
die Zeit bis dorthin!?

patrick-buchenberg, 2024-07-17:
Ich bin auch sehr daran interessiert die "Veckorzeit" zu ändern, da ich enroute zum Ballonfahren nutze und dort gerne 30 min. einstellen würde,  um das Landegebiet besser abschätzen zu können. Ich würde mich sehr freuen, diese Funktion in einer zukünftigen Version zu finden!

kebekus, 2024-07-21:
@patrick-buchenberg Die Nachricht ist angekommen. Wir werden uns kümmern. Wir sind allerdings im Moment mit unserem kleinen Team völlig am Anschlag (unter anderem, weil die Infrastruktur der FAA dauernd Probleme macht), sodass ich keine Hoffnung auf eine sehr schnelle Lösung machen kann. Die Sache wird aber nicht vergessen werden.

kebekus, 2026-10-22:
Vielen Dank für diesen Vorschlag und an @patrick-buchenberg für die ergänzende Schilderung aus der Ballonfahrt – solche Einblicke, wie Enroute im Alltag genutzt wird, sind für uns sehr wertvoll.

Nach reiflicher Überlegung habe ich mich entschieden, eine einstellbare Länge des Flugbahnvektors nicht umzusetzen. Unser Team ist klein, und wir müssen unsere Kräfte auf die Dinge konzentrieren, die für möglichst viele Pilotinnen und Piloten den größten Nutzen bringen. Ich schließe das Issue deshalb, damit es keine falschen Erwartungen weckt – es tut mir leid, dass ich hier keine bessere Nachricht habe.

Das ändert nichts daran, dass Eure Ideen absolut nachvollziehbar sind. Bitte meldet Euch auch weiterhin mit Wünschen, Fehlerberichten und Anregungen – gerade dieses Feedback aus der Praxis macht Enroute besser. Herzlichen Dank für Eure Unterstützung und allzeit gute Flüge bzw. Fahrten!

closed 2026-10-22 by kebekus