Akaflieg-Freiburg/enroute #358: Keine wetterdaten an flugplätzen. Metat taf

Ludwig2610, 2023-11-01:
Ich bekomme keine Meter und Taf Daten mehr. 
Somit keine wetterdaten an flugplätzen

kebekus, 2023-11-02:
@Ludwig2610 Welche Programmversion verwenden Sie? Und welches Gerät?

Ludwig2610, 2023-11-02:
Hallo

Samsung   Galaxy  S 20 FE   Android

Version   2.29.3 GIT    68444b92

 

MFG

 

 

Ludwig Weidinger

 

Franz-Weindler-Prommenade . 10

85435 ERDING

016094458367

 

images

kebekus, 2023-11-03:
@Ludwig2610 Danke für die Info, die mich allerdings wundert. Es gab in der jüngeren Vergangenheit Probleme, weil das "Aviation Weather Center" die Schnittstelle geändert hatte. Diese Probleme sollten aber mit Version 2.29.3 aus der Welt geschafft sein.

Können Sie bitte noch einmal überprüfen, ob die Sache vielleicht inzwischen wieder läuft? Falls nicht, können Sie mir bitte genau schreiben, was Sie machen und welche Meldungen Sie erhalten? Vielleicht können Screenshots oder ein Screencast Video helfen…

Grüße,

Stefan Kebekus.

Ludwig2610, 2023-11-03:
HalloEs geht leider nicht. Ich hänge 3 Screenshots an. Die 3 Punkte in der rechten oberen Ecke hab ich nicht. App hab ich auch neu installiert. LG und danke für die hilfe

kebekus, 2023-11-03:
@Ludwig2610 Danke für die Rückmeldung. Die Bilder sind nicht angekommen, das funktioniert per e-mail vielleicht nicht. Du kannst die Bilder entweder bei GitHub direkt in deinen/einen neuen Kommentar hochladen, oder mir alternativ direkt eine e-mail an meine Adresse schicken.

Ich bin ehrlich gesagt ein wenig ratlos. Ich verwende auf einigen Geräten ebenfalls Version   2.29.3 GIT    68444… und da funktioniert alles problemlos. Ich überlege weiter, was das Problem sein könnte…

Ludwig2610, 2023-11-08:
Okay. Hier noch mal die Fotos im Anhang.

charlylima, 2023-11-18:
Grüß Dich Stefan, 
Ich hab den selben Fehler sein ein paar Wochen. 
[image]

Metar und TAF werden von derartigen URLs abgefragt: 
* METAR:  "https://aviationweather.gov/api/data/metar?bbox=46.85,10.85,48.85,13.83"
* TAF:  "https://aviationweather.gov/api/data/taf?bbox=46.85,10.85,48.85,13.83"

Da kommt die Antwort im "raw" format und das wird dann in der Software versucht mit dem XML Parser zu dekodieren. 
`"LOWS 181750Z AUTO 15008KT 9999 FEW040 BKN100 03/02 Q1024 NOSIG\n..."`

Dann halt so, mit "format=xml": 
* METAR:  "https://aviationweather.gov/api/data/metar?format=xml&bbox=46.85,10.85,48.85,13.83"
* TAF:  "https://aviationweather.gov/api/data/taf?format=xml&bbox=46.85,10.85,48.85,13.83"

Ich frag mich bloss warum du den Fehler nicht gesehen hast? Wurden bei Dir noch alte Daten angezeigt aus weather.dat?

fbbln, 2023-11-19:
Same here:
Ich hab die 2.29.50 drauf und auch bei mir kommt nur noch die Fehlermeldung.

TTC70, 2023-11-19:
I have the same problem with Huawei mate 20 lite. Before the last update everything was working.
As far as today the Huawei mate 20 lite with the last update of Android doesn't read METAR.
Please note that Enroute v2.29.50 and Samsung Galaxy S10 works fine. 
Thanks.

charlylima, 2023-11-19:
I got a valid weather report today once with version 2.29.4. Afterwards you do not immediatly recognize if the update fails because it keeps the old data. Maybe the data format you get from aviationweather is random if you do not specify it explicitly?

kebekus, 2023-11-20:
Dear all,

I uploaded version 2.29.51 to Google Play's beta channel this morning. This version includes @charlylima 's code improvements and should fix the problem with downloading weather data. I still don't understand why the old code works on my machines and fails on the devices of so many others.

Could you please try 2.29.51 and let me know if that fixes the problem? -- Thanks.

Best,

Stefan.

fbbln, 2023-11-20:
@kebekus Positive result. It loads the weather data again.

charlylima, 2023-11-20:
Yes, for me it works.
Thank you Stefan, just in time to go flying. Weather is getting bad in Germany and it is getting more important to have the weather report available. Great Software!

kebekus, 2023-11-20:
Dear all,

After your positive feedback, I published version 2.29.51 on Google Play's main channel. Thanks to @charlylima; without your help, I would still be scratching my head.

I am closing this issue now. Please feel free to re-open if the problem persists.

Best wishes,

Stefan,

closed 2023-11-20 by kebekus