Akaflieg-Freiburg/enroute #366: Own map display is very fuzzy despite mbtiles providing zoom 11-12 level

Flying-Roger, 2024-01-06:
**Describe the bug**
Zoom levels above z10 are seemingly ignored by Enroute.
Maps using zoom levels 11, 12 becomes very fuzzy when zooming in.

**To Reproduce**
Steps to reproduce the behavior:
1. Select mbtiles file using only zoom 11 or 12 
2.  "Open with" enroute
3  Displayed map is empty
or
1. Select mebtiles files using zooms 8-12
2. "Open with" enroute
3.  Map is displayed, but becomes very fuzzy when zooming in despite providing zoom 11, 12 in the mbtiles file

**Smartphone (please complete the following information):**
 - Device: Samsung A34
 - OS: Android 14

kebekus, 2024-01-06:
@Flying-Roger Understood. I apologize for not being very responsive over the holidays … you know how it is.

Can you give me access to the MBTILES file in question?  -- Thanks.

Flying-Roger, 2024-01-06:
Hello Stefan,

 

Thank you for your fast answer.

 

I detected the fuzzy problem importing a big map prepared with MOBAC.
However, I narrowed the search preparing small mbtiles with different zoom level (only one level in each file).

Here is one with only zoom 12.

 

Kind regards,

Roger

kebekus, 2024-01-07:
@Flying-Roger EMail attachments do not work in GitHub. Please upload the file (the small and the big one) [here](https://v2202001110709105590.ultrasrv.de/nextcloud/index.php/s/t2j4fCDot4gF3Rf). Please let me know once you are done, so I can close the upload folder.

Flying-Roger, 2024-01-07:
I uploaded the two small examples (z11 and z12), as well as a bigger map (z7-16).
You will notice that Enroute only uses z7-10 of the latter, ignoring all deeper zoom levels.

kebekus, 2024-01-07:
Files received. Where in all the world are you and what region do these maps cover?

Flying-Roger, 2024-01-07:
Oops, sorry.
I thought that tiles geolocalisation would allow you to find the area.

The map with zoom level up to 16 is centered on Bütgenbach (Belgium).

N 50:24:54.00  E 006:16:35.00

Kind regards
Roger

(Envoyé de mon smartphone)

Flying-Roger, 2024-01-08:
I had a look into the source code and found out that min and max zoom levels have to be provided in the metadata.
Seemingly mbtiles generated using MOBAC do not contain this information.
I have now patched my mbtiles files and the display is no longer fuzzy.
Problem solved; maybe a note about mbtiles metadata in the manual would help.

kebekus, 2024-01-24:
@Flying-Roger Thank you for the insight, and please apologize for my late answer. I checked the [specification document](https://github.com/mapbox/mbtiles-spec/blob/master/1.3/spec.md) for the MBTILES format today. To my surprise, I found that the metadata fields "minzoom" and "maxzoom" are not mandatory. I will add a workaround code, so **Enroute** extracts "minzoom" and "maxzoom" values from the "tiles" table if they are not provided explicitly in the file.

Flying-Roger, 2024-01-24:
Thank you !

Roger

kebekus, 2024-02-13:
(Hopefully) fixed in 2.30.4, which is to appear on Google Play later today.

@Flying-Roger Thank you for your help!

closed 2024-02-13 by kebekus