Akaflieg-Freiburg/enroute #368: Change order in layers for Flight Plan

TTC70, 2024-01-14:
Hello Stefan,
I would like to propose a slight modification of the order of the lines (magenta flight plan, 5 minutes Black+white line, and rubber band) when a flight plan is active.
At the moment, when the heading (and the red rubber line) is aligned to the flight plan route (the magenta line), the three lines are perfectly aligned. 
Unfortunately (see screenshot) the 5' B&W line is below all the rest.
I believe that the importance of this B&W line is quite high, compared to the magenta line (which continues ahead the 5'.
I suggest to change the order of these 3 lines in this way:
1. magenta line, the closest to the map;
2. B&W line (above the magenta line);
3. rubber line (it is way thinner than the B&W line so it is visible even if they are aligned) above all.

P.S.
- Reading other posts about the B&W line, i believe that 5 minutes is the perfect size, then I would not change it (but it is just my opinion).
- I don't know if it happens with other devices, it would be nice to have the head line and the bottom system buttons off when EFN is working. I don't think it is a priority.
- Personally, if you think it is possible, I would give the user/pilot the opportunity to change the limit of 3 miles far from the flight plan and have the rubber band "extended" in case the pilot wants to "recover" the flight plan he/she was following originally.

[image]

kebekus, 2024-01-21:
@TTC70 I understand the issue and will try to get the ordering of layers fixed for the next release. Thanks for catching that!

kebekus, 2024-01-21:
@TTC70 Do you actually fly above the sea? With life vest and everything? What aircraft are you using?

TTC70, 2024-01-21:
Yes indeed.
With life vest, ELT and everything.
I have a Storm Century (photo attached).
It is an ultralight airplane (TOW < 600kg).

Do you need any information in regards?

kebekus, 2024-02-07:
The issue is now fixed. The bugfix will appear in v3.29.03, to be released within the next few weeks.

kebekus, 2024-02-13:
(Hopefully) fixed in 2.30.4, which is to appear on Google Play later today.

@TTC70 Thank you for your help!

closed 2024-02-13 by kebekus

TTC70, 2024-02-16:
You got it!! 🥇