Akaflieg-Freiburg/enroute #376: Maps partially visualized

TTC70, 2024-02-10:
**Describe the bug**
Only on tablet (only recently): when you go out from your country (where you can see airports, airfields, and so on) the map is partially loaded (see screenshot here below).
I have added also the screenshot of the same area from the phone just for comparison:

**To Reproduce**
1. unzoom until the "-" (minus) button is not visible anymore.
2. scroll the map outside the borders (for instance, towards another downloaded map).
3. The map is partially visible.

**Expected behavior**
Usually the basic map is visible (see screenshot from the mobile phone - Samsung Galaxy S10).

**Screenshots**
[image]

Cheers

kebekus, 2024-02-11:
@TTC70 Understood. This does not affect the app's usability, correct? I will, therefore, classify this as "enhancement" with low priority. I hope you do not mind.

TTC70, 2024-02-11:
I confirm the issue does not affect app's usability.
Funny thing, the issue related to basic map disappearing happens only then
I un zoom with the "-" (minus) button. It does not happen with "pinch and
unzoom" fingers action.

Being a beta tester I have to report it 😊😊

Ciao!

kebekus, 2024-02-11:
> Being a beta tester I have to report it 😊😊

Absolutely! Without help, we would never find all these issues.

Thanks - Stefan.

kebekus, 2024-02-13:
(Hopefully) fixed in 2.30.4, which is to appear on Google Play later today. Thanks for all your help!

closed 2024-02-13 by kebekus