Akaflieg-Freiburg/enroute #379: App always open

nicolashayes, 2024-02-16:
Hello to all the team, i have been having a very nice time with the app flying my piper cub down here in Argentina. I am a bit frustrated only when i lock my smart phone to save some battery and when i unlock the device the screen turn on but the app is closed so i have to open it again and wait to get a gps lock. I have been using and app called oruxmaps, wich is very good for trail hiking and has the option to be always on. What do you think on this??

Best regards.

kebekus, 2024-02-18:
@nicolashayes Thank you for your report! The function you describe (=app remains open) is already implemented and **should** work. If it doesn't work, then something is wrong. To help us investigate the issue, can you please tell us what phone you are using?

nicolashayes, 2024-02-25:
Hello! thanks for the reply, i may had and old version because in my last flight the app was running in background when i switch to other app, but i have a suggestion, there is a way to make the app to keep the gps fix even in the background?? 
I will upload a video to show this issue.

Thank you for developing a super app for flying!

kebekus, 2024-03-02:
I understand that the issue has resolved itself, so I am closing this report here. Feel free to re-open if problems remain.

Please keep reporting any issues that you may find!

Best wishes from the Black Forrest,

Stefan

closed 2024-03-02 by kebekus