Akaflieg-Freiburg/enroute #380: Show aerobatic boxes on map

fly-shadok, 2024-02-17:
Just noticed that aerobatic flight boxes in German airspace are not shown on the map as they are on Eisenschmidt paper maps. As these are often used for aerobatics training, it would be useful to show them even though FIS is usually notified if aerobatics take place.

**Describe the solution you'd like**
Symbols on the map marking aerobatic boxes as defined in AIP VFR ENR 1-76 to 1-85. Neither altitude limits nor operating times are necessary, they should just serve as a heads-up reminder. The AIP format seems machine-readable to me, maybe something like the NOTAM system would suffice?

**Describe alternatives you've considered**
This could also be done on the openFlightMaps level, but they don't have a feature request option as far as I see.

**Additional context**
As the Eisenschmidt aerobatics symbol can't be used obviously (and shouldn't, because it's awful), I would suggest a simple "looping" symbol, something like what I cobbled together here:

[image]

Thanks for the great work anyway!

kebekus, 2024-02-17:
@fly-shadok Thank you for the suggestion. I am supportive in principle, but I strongly feel that the data should be added to "open flightmaps" or "openAIP." 

Reason 1: Enroute currently supports 60+ countries. This is possible because we can build on the community efforts at openAIP and open flightmaps to collect the data. Adding nation-specific data sources is a significant pain and can quickly lead to more effort than our small team can handle.

Reason 2: Data added to openAIP is available worldwide and can be used in other apps, avionics devices, printed aviation maps, etc etc

@fly-shadok Please send me a private e-mail, and I can get you in touch with the relevant people at openAIP and open flightmaps. I would rather not post their e-mail addresses on this public web site.

fly-shadok, 2024-02-17:
Hallo,

ich antworte jetzt einfach mal auf diese Mail, weil ich sonst keine Adresse finde.

Hast absolut recht, am besten wärs natürlich direkt in open flightmaps/AIP, damit auch andere darauf basierende Apps die Boxen anzeigen können.

Gruß, Gunter.

On Saturday, 17. February 2024 17:28:17 (+01:00), Stefan Kebekus wrote:

> @fly-shadok Thank you for the suggestion. I am supportive in principle, but I strongly feel that the data should be added to "open flightmaps" or "openAIP."
>
> Reason 1: Enroute currently supports [...]

kebekus, 2024-03-01:
@fly-shadok I am closing this issue because we agree that the data should be added to openAIP. Once the data is available there, I will be more than happy to show it in Enroute.

closed 2024-03-01 by kebekus