Akaflieg-Freiburg/enroute #381: Support Foreflight Content Packs

kebekus, 2024-02-21:
One user wrote: "ForeFlight hat da übrigens ein sehr interessantes, offenes Format mit einem guten Bedienkonzept. Vielleicht wäre das ja auch etwas für Enroute."

mipastgt, 2024-02-21:
Hi from "One user". Here are a few links with some information about ForeFlight content packs.
https://foreflight.com/products/foreflight-mobile/user-content/content-packs/
https://foreflight.com/support/content-packs/

Interesting concepts are:

- Select a aerodrome on the map and then view all associated documentation either in a document view or also on the map.
- It can handle georeferenced PDF files.
- You can define user waypoints and also associate documentation with them.

The current version of **AIP Browser DE** already supports [trip-kits](https://mpmediasoft.de/products/AIPBrowserDE/help/AIPBrowserDE.html#_ein_trip_kit_erstellen) in this format.

kebekus, 2024-12-01:
Dear all, we considered the suggestion and eventually decided again implementing it. Reason: the data and data formats in the ForeFlight Packs to not fit well with the logic and technology of Enroute. As a minimum, we would have to include a full-blown PDF reader/converts into Enroute, which seems like a tech overkill that will expand the size of our app dramatically and will bring all sorts of performance issues.

I am sorry that I have no better news.

closed 2024-12-01 by kebekus