Akaflieg-Freiburg/enroute #382: Popup bug

2dylan1, 2024-02-26:
If i remove a checkpoint on the Route and wind window, the pop up stays there

Steps to reproduce the behavior:
1. Go to route and wind
2. Click on a waypoint to remove the waypoint 
3. Scroll down to remove 
4. See error

Im on Android 
Xiaomi Redmi note 11 pro 

App version v2.30.5

[image]

kebekus, 2024-02-26:
@2dylan1 Thank you for the report. I can reproduce the issue here and will try to come up with a fix ASAP.

Best,

Stefan.

kebekus, 2024-02-27:
@2dylan1 The issue is (hopefully) fixed in 2.30.7, which is now available for beta testing at Google Play. Can you please try the new version and let me know if the problem is resolved for you? -- Thanks!

closed 2024-03-01 by kebekus