Akaflieg-Freiburg/enroute #383: Falsche Höhenangaben wenn Flugzeug metrisch ist

Ventusx6, 2024-02-27:
**Describe the bug**
A clear and concise description of what the bug is.
Wenn in der Flugzeugangabe Meter für die vertikale Massangabe gewählt wird, dann werden Flightlevel Angaben in Meter umgerechnet. Als zB FL195 wird dann als FL 5950 m angezeigt. FL Angaben sollten NIE umgerechnet werden. 
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

kebekus, 2024-02-28:
@Ventusx6 Danke für die Nachricht. Mir ist allerdings nicht ganz klar, warum FL Angaben ihrer Meinung nach nie umgerechnet werden sollten. Können Sie mir hier helfen? -- Ich selbst bin für eine Umrechnung immer ganz dankbar, wenn ich mich im Sommer mit dem Segelflugzeug der Grenze des oberen Luftraums nähere.

Ventusx6, 2024-02-28:
Hallo Stefan

Ich nehme mal an, dass du wie folgt umrechnest:

FL100 = 10'000 Fuss = 3048 Meter. Das gilt für die Standardatmosphäre von 1013,25 hpa. Dann ist FL100 gleichgesetzt mit 3048 Meter über Meer.

Wenn du nun in einem Tiefdruckgebiet auf einem Flugplatz startest und am Höhenmesser QNH einstellst, dann musst du bei der Umrechnung einen Korrekturfaktor von ca 10 Meter pro 1 hpa berücksichtigen.

Wenn du am Flugplatz ein QNH von 1000 hpa hast und das so einstellst, dann beträgt die Korrekturhöhe gegenüber FL ca 130 Meter (13hpa * 10 Meter). Wenn du nun auf deine gerechneten 3048 Meter über Meer steigst, dann befindets du dich bereits ab ca 2'918 Meter über Meer auf FL100. Steigts du weiter auf deine gerechneten 3048 Meter über Meer, dann bist du 130 Meter zu hoch, das heisst du hast eine Luftraumverletzung gemacht.

Anders verhält es sich, wenn du während dem Flug deinen Höhenmesser auf Standardhöhe 1013,2 hpa einstellst. Aber ich kenne keine Segelflugpiloten, die das machen.

Die neuen Instrumente (bsp LX-Geräte) können gleichzeitig Meter über Meer und den FL (umgerechnet auf 1013,2 hpa) anzeigen.

Also habe ich einen Luftraum von zB 4500 Meter, dann gucke ich auf die Meterangabe im LX-Instrument, habe ich einen Luftraum mit FL130, dann gucke ich auf die FL-Angabe in LX-Instrument. Wird mir nun aber die FL-Angabe in Metern angezeigt (wie in der APP), dann nützt mir (und den meisten Piloten) das gar nichts, da ich dann die Korrektur von 1hp = 10 Metern einrechnen muss oder ich muss den Höhenmesser auf 1013,2 hpa umstellen. Sobald ich dann wieder einen Luftraum in AMSL habe muss ich den Höhenmesser wieder auf QNH umstellen.

Wichtig zu wissen ist folgendes: Starte ich in einem Hochdruckgebiet kann ich mich auf die Meterangabe verlassen (ich fliege dann einfach zu tief was kein Problem ist). Starte ich in einem Tiefdruckgebiet, dann muss ich die Korrektur anwenden, da ich ansonsten eine Luftraumverletzung begehe.

Für mich ist ein System, das die Flugfläche in Metern umrechnet ein no-go. Ich fliege zeitlebens mit QNH rum. Ich frage mich wie viele Piloten bei deiner Umrechnung ahnungslos in einen Luftraum einfliegen, für den sie eigentlich eine Bewilligung bräuchten. Eine Luftraumverletzung ist eine Verletzung, auch wenn es sich nur um 100 Meter handelt. Und jede Luftraumverletzung schadet allen Piloten, da dann die Autoritäten grössere Margen einbauen und die Lufträume einfach senken. Wenn die DFS bsp den Luftraum «C» auf FL90 senken, diesen aber erst ab FL100 bewirtschaften, können ihnen keine Piloten in die Quere kommen, die mit QNH rumfliegen und FL auf Meter umrechnen.

Reinhold Müller

_____________________
Reinhold Müller
Nellweg 17b
5018 Erlinsbach

Mobile:    +41 79 692 29 24
Fax:           +41 62 844 25 91

kebekus, 2024-03-05:
Lieber Reinhold,

danke für die Erklärungen. Wir haben das im Kreis der Entwickler diskutiert, können Deiner Erklärung aber nicht folgen. Wir haben daher beschlossen, die App an dieser Stelle nicht zu ändern. Die Gründe sind die folgenden.

1) Oberhalb der Übergangshöhe ist der Höhenmesser verpflichtend auf den Standarddruck einzustellen; das ist in der AIP Deutschland ENR 1-14 klar festgelegt:

https://aip.dfs.de/BasicVFR/2024FEB08/pages/804d7d850155c88608f8659768fe0846.html

Für alle anderen mir bekannten Länder gilt ähnliches. Obwohl sich die App hauptsächlich an Motorflieger richtet, bin ich selbst auch Segelflieger und kann mit Sicherheit sagen, dass das in unserem Verein auch so praktiziert wird -- wir fliegen im Sommer im Schwarzwald oft bis FL95; im Winter gibt es bei uns machmal Welle. In unserem Motorflugzeugen und Überland-Segelflugzeugen haben wir aus diesem Grund auch zwei Höhenmesser eingebaut. Wir denken, dass es nicht Aufgabe der App sein kann, regelwidriges Verhalten Einzelner zu unterstützen.

2) Bezeichnungen wie "FL 5950 m" sind üblich und werden auch in den offiziellen ICAO-Karten der DFS verwendet (genauer: in der metrischen Ausgabe für Segelflieger). Durch das vorstellte "FL" ist klar, dass es sich um eine Flugfläche handelt. 

Ich verstehe, dass das nicht die Antwort ist, die Du hören wolltest. Ich bitte um Verständnis, dass wir Deinem Vorschlag aus den genannten Gründen nicht folgen.

Grüße,

Stefan.

closed 2024-03-05 by kebekus

Ventusx6, 2024-03-20:
Guten Morgen Stefan

Nun habe ich doch noch einen Fehler gefunden. Das Programm macht aus Airspace Classe E in der Uebersetzung Airspace Classe EAST.

Gruss

***@***.******@***.***
Reinhold Müller

_____________________
Reinhold Müller
Nellweg 17b
5018 Erlinsbach

Mobile:    +41 79 692 29 24
Fax:           +41 62 844 25 91