Akaflieg-Freiburg/enroute #385: Import von geoTIFF scheitert

RobeKrau, 2024-03-01:
Beim Versuch, ein GeoTIFF in die VAC-Liste zu importieren, wird ein Fehler angezeigt und der Prozess bricht ab.

Öffne den Dateimanager
wähle die zuvor im Verzeichnis Download abgelegte Datei 
starte "öffne mit - Enroute".

Screenshot der Fehlermeldung:
[image]

- Enroute V2.29.52 
- Device: Samsung Galaxy Note 10+ (SM-N975F)
- OS: Android 7.1.2

Das GeoTIFF wurde erzeugt mit AIP Browser DE V1.1.4, siehe Anhang (Endung für den Upload hier geändert auf .TXT, bitte zum Testen wieder ändern in .TIFF). 
[EDKD Altena-Hegenscheid 1-geo_TIFF.txt](https://github.com/Akaflieg-Freiburg/enroute/files/14458377/EDKD.Altena-Hegenscheid.1-geo_TIFF.txt)
Es wird mit üblichen Grafik-Betrachtern (unter Windows 10) korrekt angezeigt.

kebekus, 2024-03-02:
@RobeKrau Dabke für die Nachricht. Das Problem tritt auf meinen eigenen Geräten nicht auf. Ich frage mich, ob die Version 2.29.52 nicht vielleicht einfach zu alt ist -- die Funktionalität ist ja recht neu. Können Sie bitte auf die aktuelle Version (=2.30.8) update und den Import der Datei nocheinmal testen?

Grüße aus Freiburg,

Stefan Kebekus,

RobeKrau, 2024-03-02:
@kebekus Vielen Dank für die prompte Reaktion!

Ich habe EFN V2.30.8 aufgespielt und es mit einem anderen Androiden (OS V10) versucht. Leider bleibt es bei der Fehlermeldung.
Mir fällt auf, dass die TIFF, wenn man sie mit den üblichen Grafikbetrachtern unter Android (z.B. "Galerie") öffnet, nur eine schwarze Fläche erzeugt (im Gegensatz zur Windows-Umgebung, wie im Eingangs-Post beschrieben).

Grüße aus dem Sauerland,

Robert

kebekus, 2024-03-02:
@RobeKrau Lieber Robert, das ist jetzt tatsächlich interessant. Offenbar hängt das Problem ab, welche Version der libtiff mit den unterschiedlichen Android-Versionen ausgeliefert wird. Das Bild, das Du verwenden möchtest, ist offenbar für viele Androide nicht lesbar und funktioniert deshalb auch in Enroute nicht. Auf anderen Platformen läuft alles ganz problemlos. Ein Test mit iOS zeigt ein korrektes Vorschaubild, kann aber das tatsächliche Bild nicht laden.

Ich sehe zwei Problemquellen. Zum einen lässt das Format TIFF lässt irrsinnig viele Untervarianten zu, die von praktisch keinem einzigen Programm alle abgedeckt werden können. Vielleicht ist Deine Datei auch beschädigt. Was das Problem jetzt genau bei Dir ist, habe ich noch nicht herausgefunden, das wäre wahrscheinlich auch recht zeitaufwändig.

Eine gute Lösung habe ich nicht. Wir können im Prinzip zwar eine eigene Version von libtiff ausliefern, die weitere Formatvarianten unterstütze, aber der Wartungsaufwand ist praktisch nicht darstellbar.

Zwei Fragen zum Schluss:

- Wo kommt das Bild denn her?

- Hast Du andere Bilder, die korrekt in der Galerie angezeigt werden und in Enroute auch funktionieren?

Grüße,

Stefan.

RobeKrau, 2024-03-02:
@kebekus Hallo, Stefan,

das GeoTIFF wurde erzeugt mit AIP Browser DE V1.1.4 (wie im Eingangspost erwähnt).

Es gibt schon wieder Neuigkeiten:
Auf einem Android-Gerät mit OS V14 wird das (identische) geoTIFF zwar auch nicht angezeigt in "Google Foto" oder "Galerie", es wird aber klaglos von EFN eingelesen und erscheint auch korrekt, wenn in EFN "Anflugkarten" gewählt wird! 🙂

Vielleicht kennt jemand einen TIFF-Konverter, der die von Michael Paus Software erzeugten Dateien in ein Android 10-geeignetes Format umrechnet? Oder Michael wählt direkt ein passendes anderes Format?

👋 Robert

RobeKrau, 2024-03-04:
Hier noch eine etwas andere Fehlermeldung von einem anderen Endgerät:
[image]

mipastgt, 2024-03-04:
Den beschriebenen Fehler kann ich übrigens auch reproduzieren.
Auf meinem alten Samsung Galaxy Tab S4 und Android 10 geht es.
Auf meinem nicht ganz so alten Xiaomi Mi 11 Lite 5G und Android 11 geht es nicht.
Enroute Flight Navigation ist in beiden Fällen v2.30.8.
[image]

RobeKrau, 2024-03-04:
Der Import des von @mipastgt generierten Trip Kits [trip-kit-config-EDKD-charts-only.json.zip](https://github.com/Akaflieg-Freiburg/enroute/files/14480666/trip-kit-config-EDKD-charts-only.json.zip) hat leider auch nicht funktioniert:
[image]

kebekus, 2024-03-04:
@mipastgt @RobeKrau Danke für Eure vielen Beiträge! Ich bin im Moment leider für einige Tage außer Gefecht, weil ich auf Dienstreise in Lausanne bin und alle Hände voll zu tun habe. Vermutlich wird es ein wenig dauern, bis ich sinnvoll am Gespräch teilnehmen kann. Bis dahin bitte ich um etwas Geduld!

mipastgt, 2024-03-04:
@RobeKrau Da lag jetzt bei Dir ein Missverständnis vor. Der Link von mir oben beinhaltet kein Trip-Kit sondern eine Konfigurationsdatei um eines zu erzeugen. D.h., Du musst das ZIP-File auspacken und die Konfigurationsdatei darin dann mit dem AIPBrowserDE unter Settings->Trip-Kit->Select/Export auswählen. Dann wird das Trip-Kit erzeugt. Ich habe aber hier mal tatsächlich das entsprechend erzeugte Trip-Kit angehängt, das man direkt in Enroute laden können sollte. [EDKD.zip](https://github.com/Akaflieg-Freiburg/enroute/files/14486411/EDKD.zip)

pakthan, 2024-03-07:
Hallo!
Da ich gerade tolle Hilfe im AIP Browser GIT bekommen habe, wollte ich an dieser Stelle kurz eine Beobachtung melden.

Ich wollte ein Trip-Kit ZIP importieren, welches die gesamten deutschen Anflugkarten beinhaltete.
Wenn ich auf einem Android V12 Gerät die "Öffnen mit..." Funktion von Dateibrowsern oder Total Commander nutze, crashed EFN mit dem hier so oft genannten "Der Typ der Datei XYZ wird nicht erkannt".

Verwendet man die Import Funktion unter Bibliothek-VAC, kann man auch die ZIP Dateien auswählen und EFN bietet sofort den Import an. 

Vielleicht hilft diese Info ja den einem oder anderen.

RobeKrau, 2024-03-07:
@pakthan Danke für den Tipp mit der Importfunktion im VAC-Bereich, diese Option war mir überhaupt noch aufgefallen.

Erstaunlich: das Android-Gerät, das bisher den Import von geoTIFFS verweigerte (übrigens nicht mit einem Crash, sondern mit einer "ordentlichen" Fehlermeldung 😉), verweigert seit gestern nicht mehr.
Die Ursache kenne ich nicht. Seit der Thread-Eröffnung und gestern habe ich nebenbei auf einem Nextcloud-Problem herumgekaut. Das führte u.a. dazu, dass auf dem Gerät Google Chrome und Android System WebView neu installiert wurden. Anscheinend hat das auch das oben beschriebene Problem mit den geoTIFFs gelöst 🤷🏻‍♂️.

mipastgt, 2024-03-07:
Es geschehen noch Zeichen und Wunder, aber wie verpackt man das jetzt in eine Handlungsanweisung für jemanden der evtl. das gleiche Problem hat?

kebekus, 2024-04-05:
Liebe Leute, soweit ich das erkennen kann, hat sich das Problem in irgendeiner Weise von selbst gelöst. Ich schließe diesen Eintrag daher jetzt. Bitte gern wieder öffnen, wenn ich hier einen Fehler gemacht haben sollte.

@RobeKrau Bitte gern weitere Probleme melden. Es gibt so viele mögliche Fehler, dass wir niemals in der Lage sein werden, alle Schwierigkeiten selbst zu finden…

closed 2024-04-05 by kebekus