Akaflieg-Freiburg/enroute #387: Smaller Title bar while in VAC mode

pakthan, 2024-03-07:
Hi!
Thank you very  much for your wonderful App! It is really a great help and useful addition to the daily flying tasks!

I use a 6" android phone strapped to my wrist, so I don't need a suction mount or similar in small planes.
When I use your great VAC mode, I get this petrol colored Title Bar, which consumes a lot of space, when the phone is oriented in landscape mode.
Due to the longer aspect ratio this is a typical problem of phones instead of tablets.

Maybe the bar could be reduced to a button? Or just one corner?

Just a suggestion.

kebekus, 2024-03-08:
@pakthan Thanks for your report. We will consider the suggestion. However, as Enroute is gaining a larger and larger user basis, we are flooded with reports and suggestions. We will therefore need substantial time before we can come back to your suggestion. Your understanding is highly appreciated!

kebekus, 2024-08-24:
Thanks again. We looked into the request and decided not to consider the issue. Reason: GUI object sizes are defined by Google's Material Design system and predefined in the computer libraries we use. Changing those is difficult and quite error-prone. Since we are a small team, we need to prioritize our efforts strictly, so all we can do is ask for your understanding.

If you are willing to implement the feature yourself (or if you know anybody), then please be in touch.

closed 2024-08-24 by kebekus