Akaflieg-Freiburg/enroute #388: IOS Import/Teilen geo.tiff VAC -> Enroute wird nicht angezeigt

tittelm, 2024-03-07:
Danke für diese wunderbare App! Ihr leistet großartige Arbeit!

Beim Versuch eine geo.tiff Anflugkarte in "Enroute" zu importieren wird mir beim "Teilen" der Datei die App "Enroute" von IOS Iphone 13 nicht vorgeschlagen. Normalerweise wird dem Dateiformat zugeordnet eine App vorgeschlagen.

1. Datei bereitgestellt durch: AIP Browser DE -> Format geo.tiff
2. An das Iphone gesendet per Mail/Dropbox/Icloud/Google drive (Mehrere Möglichkeiten ausprobiert)
3. Datei auf Iphone aufgerufen und "Teilen" ausgewählt
4. Vorgeschlagene Apps: große Auswahl unter anderem auch Foreflight, Air Navigation und Garmin -> "Enroute" fehlt auf der Liste
5. Versuch die "Teilen" Vorschlagliste anzupassen (direkt auf der "Teilen" Oberfläche) und zusätzlich unter -> Einstellungen/ Siri&Suchen/ Enroute
6. Keine Möglichkeit "Enroute" auszuwählen.

 - IOS 17.4
 - Datei über AIP Browser -> geo.tiff
 - Iphone 13
 - Enroute v2.30.8

Vielen Dank für Eure Hilfe!

Alles Liebe Moritz

schneisi, 2024-03-07:
Hallo Moritz,

danke für den Report. Ich werde mir das anschauen.

pakthan, 2024-03-08:
Hi!
Ich habe ein ähnliches Problem auf Android gehabt (Öffnen mit...) 
Bei mir Funktioniert es, wenn man in der EFN App unter Bibliothek / Karten/VAC den "Importieren" Button nutzt.
So bekomme ich TIFF und Tripkit ZIP geöffnet. Vielleicht klappt es ja so auch auf dem Iphone

tittelm, 2024-03-08:
> Hi! Ich habe ein ähnliches Problem auf Android gehabt (Öffnen mit...) Bei mir Funktioniert es, wenn man in der EFN App unter Bibliothek / Karten/VAC den "Importieren" Button nutzt. So bekomme ich TIFF und Tripkit ZIP geöffnet. Vielleicht klappt es ja so auch auf dem Iphone

Hi, danke für den Tipp. Das habe ich auch versucht aber auf dem IPhone ist in der  EFN App unter Bibliothek/Karten/3Dots der „Importieren“ Button ausgegraut und leider nicht anwählbar. Liebe Grüße, Moritz

schneisi, 2024-03-15:
Dauerte jetzt doch leider etwas länger, da ich beruflich gerade viel unterwegs bin.
Rollout der 2.30.9 ist angelaufen, diese enthält den Fix.

Gerne testen und kurz Bescheid geben ob es jetzt wie gewünscht funktioniert. Danke

> Hi! Ich habe ein ähnliches Problem auf Android gehabt (Öffnen mit...) Bei mir Funktioniert es, wenn man in der EFN App unter Bibliothek / Karten/VAC den "Importieren" Button nutzt. So bekomme ich TIFF und Tripkit ZIP geöffnet. Vielleicht klappt es ja so auch auf dem Iphone

In Enroute einen Dateibrowser öffnen und auf diese Weise einen Import starten, ist auf iOS zum aktuellen Zeitpunkt nicht möglich.

tittelm, 2024-03-15:
Top funktioniert bestens!
Getestet: AIP (Export Anflugkarte geo.tiff) -> gesendet via Mail ans IPhone -> direkt aus Mail "Teilen" -> Enroute wird App-auswahlmöglichkeit angezeigt
:-)
Danke!

closed 2024-03-15 by schneisi