Akaflieg-Freiburg/enroute #389: App unstable / unresponsive on Pixel 5

xxDUxx, 2024-03-09:
Pixel 5 with stock Android 14, bluetooth and Wifi off, no other apps running. 950MB trip kit installed. 
Enroute works normally about 50% of the time. Touch scrolling usually stops working after a minute or so. About once every hour, the app freezes completely.

kebekus, 2024-03-09:
@xxDUxx Thanks for the report. This problem is new to us, we have never observed the behaviour on any of our devices. I have to admit that I am a little clueless about possible causes.

To begin the investigatio, could your please help us with the following data? -- Thanks.

- What version of Enroute do you use exactly?
- How many maps do you have installed?
- Does the problem get better when you uninstall the approach charts?
- Do you use the waypoint library?

Best,

Stefan.

xxDUxx, 2024-03-10:
- Enroute Version 2.30.8
- Only German map and FLARM Database installed
- I tried both with and without approach charts, no difference
- no waypoints, no routes, no aircraft library
The problem occurs both in the air an on the ground.

kebekus, 2024-03-11:
@xxDUxx Danke für die Nachricht, die mich allerdings zunehmend ratlos lässt.

Nächster Versuch: Enroute kann zu Debug-Zwecken Systeminformationen extrahieren. Kannst Du das folgende tun, wenn es wieder Probleme gegeben hat:

- Enroute ggfs neu starten
- Hauptmenü aufmachen, dann Information/über Enroute.
- Auf der Seite "Über EFN" den Reiter "System" wählen.
- Dort unten den Knopf "Info teilen"
- Eine geeignete App wählen, um die Information weiterzuleiten (E-Mail, Messenger, ...)

Ich sage schon mal "danke".

xxDUxx, 2024-03-12:
<h3>App</h3>
<table>
<tr></tr>
<tr><td>Enroute Version<td><td>2.30.8<td></tr>
<tr><td>GIT <td><td>8ffcc6ca<td></tr>
<tr><td>Qt<td><td>6.6.2<td></tr>
</table><br>
<h3>System</h3>
<table>
<tr></tr>
<tr><td>Build ABI<td><td>arm64-little_endian-lp64<td></tr>
<tr><td>Build CPU<td><td>arm64<td></tr>
<tr><td>Current CPU<td><td>arm64<td></tr>
<tr><td>Kernel Type<td><td>linux<td></tr>
<tr><td>Kernel Version<td><td>4.19.278-g7b0944645172-ab10812814<td></tr>
<tr><td>System Name<td><td>Android  (14.0)<td></tr>
<tr><td>System Type<td><td>android<td></tr>
<tr><td>System Version<td><td>14.0<td></tr>
</table><br>
<h3>Device</h3>
Google redfin (Pixel 5)<h3>System Log</h3>
<pre>
--------- beginning of main
03-12 02:56:55.781 32391 32391 W Qt A11Y : AccessibilityNodeInfo with empty contentDescription: -2147483208
03-12 02:56:55.797 32391 32391 W Qt A11Y : AccessibilityEvent with empty description
03-12 02:56:55.817 32391 32391 W Qt A11Y : AccessibilityNodeInfo with empty contentDescription: -2147483208
03-12 02:56:55.849 32391 32391 W Qt A11Y : AccessibilityEvent with empty description
03-12 02:56:55.850 32391 32391 W Qt A11Y : AccessibilityNodeInfo with empty contentDescription: -2147483208
03-12 02:56:55.884 32391 32391 W Qt A11Y : AccessibilityEvent with empty description
03-12 02:56:55.885 32391 32391 W Qt A11Y : AccessibilityNodeInfo with empty contentDescription: -2147483208
03-12 02:56:55.916 32391 32391 W Qt A11Y : AccessibilityEvent with empty description
03-12 02:56:55.918 32391 32391 W Qt A11Y : AccessibilityNodeInfo with empty contentDescription: -2147483208
03-12 02:56:55.979 32391 32391 W Qt A11Y : AccessibilityEvent with empty description
03-12 02:56:55.986 32391 32391 W Qt A11Y : AccessibilityNodeInfo with empty contentDescription: -2147483208
03-12 02:56:56.015 32391 32391 W Qt A11Y : AccessibilityEvent with empty description
03-12 02:56:56.015 32391 32391 W Qt A11Y : AccessibilityNodeInfo with empty contentDescription: -2147483208
03-12 02:56:56.016 32391 32391 W Qt A11Y : AccessibilityEvent with empty description
03-12 02:56:56.016 32391 32391 W Qt A11Y : AccessibilityNodeInfo with empty contentDescription: -2147483208
03-12 02:56:56.017 32391 32391 W Qt A11Y : AccessibilityEvent with empty description
03-12 02:56:56.017 32391 32391 W Qt A11Y : AccessibilityNodeInfo with empty contentDescription: -2147483208
03-12 02:56:56.018 32391 32391 W Qt A11Y : AccessibilityEvent with empty description
03-12 02:56:56.029 32391 32391 W Qt A11Y : AccessibilityNodeInfo with empty contentDescription: -2147483208
03-12 02:56:56.048 32391 32391 W Qt A11Y : AccessibilityEvent with empty description
03-12 02:56:56.048 32391 32391 W Qt A11Y : AccessibilityNodeInfo with empty contentDescription: -2147483208
03-12 02:56:56.049 32391 32391 W Qt A11Y : AccessibilityEvent with empty description
03-12 02:56:56.062 32391 32391 W Qt A11Y : AccessibilityNodeInfo with empty contentDescription: -2147483208
03-12 02:56:56.081 32391 32391 W Qt A11Y : AccessibilityNodeInfo with empty contentDescription: -2147483208
03-12 02:56:56.082 32391 32391 W Qt A11Y : AccessibilityNodeInfo with empty contentDescription: -2147483208
03-12 02:56:56.113 32391 32391 I Choreographer: Skipped 48 frames!  The application may be doing too much work on its main thread.
03-12 02:56:56.129 32391 32413 I OpenGLRenderer: Davey! duration=817ms; Flags=0, FrameTimelineVsyncId=131635282, IntendedVsync=573652478118433, Vsync=573653277744433, InputEventId=173683955, HandleInputStart=573653289842546, AnimationStart=573653290379317, PerformTraversalsStart=573653290381348, DrawStart=573653290486453, FrameDeadline=573652498618433, FrameInterval=573653289476505, FrameStartTime=16658875, SyncQueued=573653290557234, SyncStart=573653290601140, IssueDrawCommandsStart=573653290680307, SwapBuffers=573653291554057, FrameCompleted=573653295805411, DequeueBufferDuration=30260, QueueBufferDuration=300469, GpuCompleted=573653295805411, SwapBuffersCompleted=573653292309317, DisplayPresentTime=0, CommandSubmissionCompleted=573653291554057, 
03-12 02:56:56.782 32391 32412 W enroute_flight_navigation: qrc:/qml/dialogs/WaypointDescription.qml:394:13: QML ColumnLayout: Layout polish loop detected for QQuickColumnLayout(0x70341b8310, id="co", parent=0x7074135af0, geometry=0,0 244x203). Aborting after two iterations.
03-12 02:56:56.793 32391 32391 W Qt A11Y : AccessibilityEvent with empty description
03-12 02:56:56.811 32391 32391 W Qt A11Y : AccessibilityNodeInfo with empty contentDescription: -2147483214
03-12 02:56:56.865 32391 32391 W Qt A11Y : AccessibilityEvent with empty description
03-12 02:56:56.866 32391 32391 I ImeTracker: de.akaflieg_freiburg.enroute:cd25ff66: onRequestHide at ORIGIN_CLIENT_HIDE_SOFT_INPUT reason HIDE_SOFT_INPUT
03-12 02:56:56.866 32391 32391 I ImeTracker: de.akaflieg_freiburg.enroute:cd25ff66: onFailed at PHASE_CLIENT_VIEW_SERVED
03-12 02:56:56.886 32391 32391 W Qt A11Y : AccessibilityEvent with empty description
03-12 02:56:56.887 32391 32391 W Qt A11Y : AccessibilityNodeInfo with empty contentDescription: -2147483214
03-12 02:56:56.912 32391 32391 W Qt A11Y : AccessibilityEvent with empty description
03-12 02:56:56.934 32391 32391 W Qt A11Y : AccessibilityNodeInfo with empty contentDescription: -2147483214
03-12 02:56:57.980 32391 32391 W Qt A11Y : AccessibilityEvent with empty description
03-12 02:56:57.981 32391 32391 I ImeTracker: de.akaflieg_freiburg.enroute:dc33ea6d: onRequestHide at ORIGIN_CLIENT_HIDE_SOFT_INPUT reason HIDE_SOFT_INPUT
[... cut ...]
03-12 02:57:33.295 32391 32391 I ImeTracker: de.akaflieg_freiburg.enroute:1b8c8b05: onRequestHide at ORIGIN_CLIENT_HIDE_SOFT_INPUT reason HIDE_SOFT_INPUT
03-12 02:57:33.295 32391 32391 I ImeTracker: de.akaflieg_freiburg.enroute:1b8c8b05: onFailed at PHASE_CLIENT_VIEW_SERVED
03-12 02:57:34.595 32391 32391 I ImeTracker: de.akaflieg_freiburg.enroute:2a504fae: onRequestHide at ORIGIN_CLIENT_HIDE_SOFT_INPUT reason HIDE_SOFT_INPUT
03-12 02:57:34.595 32391 32391 I ImeTracker: de.akaflieg_freiburg.enroute:2a504fae: onFailed at PHASE_CLIENT_VIEW_SERVED

</pre>

kebekus, 2024-03-13:
@xxDUxx Danke für die Daten. Ich bin allerdings im Moment immer noch ziemlich ratlos...

xxDUxx, 2024-07-24:
I may have found a solution: All works well when i disable "gesture navigation" in Android settings. The "AccessibilityEvent with empty description" entries in the system log also disappear.

kebekus, 2024-09-16:
Another user had reported the problem and confirmed that disabling gesture navigation in Android settings solved the problem for him. 

I think I should try to build Qt for Android without accessibility settings. That will, however, take a while.

kebekus, 2026-10-18:
Thank you very much, @xxDUxx, for your patience and for the careful investigation. Finding the link to Android's gesture navigation was a real breakthrough! I believe the problem has been solved in the meantime, so I am closing this issue now.

Please keep reporting any problems you find. Your feedback makes a real difference. If the app becomes unstable or unresponsive again on your Pixel 5, just re-open this issue and we will continue from here.

closed 2026-10-18 by kebekus