Akaflieg-Freiburg/enroute #391: Support more GeoTIFF formats

kebekus, 2024-03-15:
Michael Paus wrote:

mir ist noch etwas aufgefallen. Ich kann mit Enroute keine Geo-TIFFs mehr importieren,
die ich mit meiner neuesten Softwareversion erzeugt habe. Als Grund vermute ich eine
Umstellung bei mir. Ich schreibe die Geoinformationen jetzt in den TIFF-Tag
GeoTIFFTagSet.TAG_MODEL_TRANSFORMATION.

Dieser hat als Parameter eine 4x4 affine Transformationsmatrix, von dem ich allerdings
nur den ebenen 3x3 Teil nutze. Der Vorteil dieser Methode ist, dass ich damit auch gedrehte
Charts abbilden kann und auch eine Ausgleichsrechnung machen kann.

val modelTransformation = doubleArrayOf(
    modelMatrix[0],  modelMatrix[1], 0.0, modelMatrix[2],
    modelMatrix[3],  modelMatrix[4], 0.0, modelMatrix[5],
    0.0,             0.0           , 0.0,            0.0,
    0.0,             0.0           , 0.0,            1.0,
)

Mit QGIS kann ich die neuen Tiffs problemlos lesen. Beispiel siehe Anhang. Es wäre schön,
wenn Du diese Art der Geoinformation zusätzlich auch unterstützen würdest.

Ich habe mir auch nochmal die GeoTIFF Spezification angesehen.
[<http://geotiff.maptools.org/spec/geotiff2.6.html>](http://geotiff.maptools.org/spec/geotiff2.6.html)

Danach gibt es folgende Möglichkeiten die Geoinformationen zu spezifizieren:

1. Ein raster->model tiepoint pair + (ScaleX, ScaleY, ScaleZ) = Keine Drehung/Scherung möglich

2. Zwei raster->model tiepoint pairs = Keine Drehung/Scherung möglich

3. Drei raster->model tiepoint pairs = Allgemeine, affine Transformation ist eindeutig definiert.

4. Mehr als drei raster->model tiepoint pairs = Verhalten undefiniert. Approximation einer affinen Transformation z.B. über least-square Ansatz möglich.

5. Direkte Angabe der Affinen Transformation. = Allgemeine, affine Transformation ist eindeutig definiert. Es wäre sogar eine perspektivische Transformation möglich, da 4x4 Matrix.

Welche dieser Methoden unterstützt Enroute denn momentan eigentlich?

Ich hatte bisher 1. verwendet und nehme jetzt 5. wegen der Flexibilität (allerdings nur die 2D Variante).

---

Des weiteren habe ich Probleme bei gedrehten Bildern festgestellt, die über ein Trip-Kit
importiert wurden. Diese sind in der Darstellung völlig verzerrt. Es sieht so aus, als würdest
Du bei der Platzierung der Bilder auf der Karte nur den linken/oberen und den rechten/unteren
Punkt betrachten aber nicht die beiden anderen Punkte auch. (Siehe screenshot_tripkit.png)

[image]

[image]

kebekus, 2024-03-15:
Lieber Michael,

danke für den Hinweis! Ich schaue mir die Sache an, werde aber ein paar Tage brauchen, weil ich im Moment noch beruflich im Ausland bin.

Ich muss ehrlicherweise sagen, dass ich nur das absolute Minimum implementiert habe, das nötig war um die Beispiel-GeoTIFFs, die ich damals hatte, korrekt zu lesen. Daher ist es vielleicht nicht sehr überraschend, dass nichts mehr läuft, sobald Du andere GeoTIFF-Features verwendest.

Der derzeitige Code unterstützt die Transformationen garantiert nicht. Ich denke aber, dass ich die Funktionalität aber zeitnah nachliefern kann. Ich programmiere mal was und melde mich, vielleicht schon in der nächsten Woche.

Grüße,

Stefan.

kebekus, 2024-03-15:
@schneisi Ich habe Dich hinzugenommen, weil bestimmt auch iOS-Nutzer nach dem Problem fragen werden...

mipastgt, 2024-03-15:
Wenn es kurzfristig irgendwie hilft, dann könnte ich bei mir auch auf die Methode 3 (Drei raster->model tiepoint pairs) umstellen. Langfristig ist das aber auch keine Lösung, denn GeoTIFFs können ja aus beliebigen Quellen kommen und dann hat man auch keine Kontrolle darüber welche der möglichen Methoden zur Beschreibung der Georeferenzierung verwendet wird. Ausserdem ist bei den drei Tiepoints auch nicht gesagt, dass diese immer mit den Eckpunkten korrespondieren. Im allgemeinen Fall müßte man aus diesen Tiepoints sowieso die affine Transformationsmatrix berechnen.

kebekus, 2024-03-19:
Enroute is now internally able to read GeoTIFFs where the transformation is specified in one of the following ways:

- via a ModelTransformationTag
- via a combination of ModelPixelScaleTag and ModelTiepointTag

The changes are not user visible yet, because Enroute cannot presently handle VACs with that info; I will implement that next.

mipastgt, 2024-03-21:
I have now extended **AIP Browser DE** to be able to export pre-transformed chart images. (I explicitly say pre-transformed and not pre-rotated because a general affine transform can also have a shear component.) This works nicely with Enroute:
[image]
In the long run it would still be better if Enroute could handle all affine transforms because the images a smaller and may come from different sources which don't provide a pre-transform feature.

I have observed a problem with the zoom level though which I am going to report in a separate issue.

kebekus, 2024-03-21:
@mipastgt Thank you for the fantastic new feature. I hope that Enroute will learn to handle all affine transforms within the next two weeks… but as usual, I cannot give any guarantees.

kebekus, 2024-03-23:
Update: GeoTIFF reader can handle affine transformations now. However, we still need to implement support for that in the VAC handling code of Enroute. A major re-write of the VAC library class is required.

kebekus, 2024-04-09:
Implemented in 2.30.13. Thanks to everybody!

closed 2024-04-09 by kebekus