Akaflieg-Freiburg/enroute #404: Add IP address of "PowerFlarm Eagle" (lxnavigation) 192.168.4.1 to the list of traffic data providers or make the IP / Port user configurable

richj1000, 2024-04-07:
The "Powerflarm Eagle" manufactured by lxnavigation uses the IP address 192.168.4.1 to provide flarm data.

Please add the IP address to the list of traffic Data providers :
 >>>   addDataSource( new Traffic::TrafficDataSource_Tcp(QStringLiteral("192.168.4.1"), 2000, this));

Thank you
Richard

kebekus, 2024-04-07:
@richj1000 Thank you for your report. I will implement the feature in the next few days and come back to you.

kebekus, 2024-04-09:
@richj1000 Dear Richard, the data source has been added to version 2.30.14, available on Google Play [but not yet on the Apple App Store]. Please confirm that everything works for you.

Also, I'd be interested to know the precise version of your "PowerFlarm Eagle" device -- I keep a list of devices that are known to work with Enroute.

Thanks again!

Stefan.

kebekus, 2024-04-13:
@richj1000 Is the problem solved for you?

kebekus, 2024-04-16:
@richj1000 Danke für die Nachricht. Ich werde ein wenig Zeit brauchen, um mir die Sache anzusehen, denn ich bin im Moment 100% damit beschäftigt, Bluetooth-Verbindungen zu implementieren […was Eure Probleme auch lösen würde…].

Zwei Fragen:

- Funktioniert der NMEA-Datentransport per WLAN denn mit anderen Apps?
- Wo seit ihr eigentlich geographisch? Falls ihr in der Nähe seit, ist es mit Sicherheit am einfachsten, wenn wir uns kurz treffen…

Grüße,

Stefan.

richj1000, 2024-04-22:
Hallo Stefan,
 
mein Trike steht in Eutingen im Gäu - als nicht weit von Freiburg - dort dürfen allerdings keine Trikes landen ... sonst würde ich sehr gerne vorbei kommen.
 
Vielen Dank für Deine klasse Arbeit!
Grüße
  Richard

kebekus, 2024-04-28:
Lieber Richard,

wenn ich richtig sehe, kann ich mir unserem Falken oder unserer Katana aber in Eutingen landen? Dann machen wir es anders herum! Ich werde aber ein wenig Zeit brauchen, weil ich im Moment 100% damit ausgelastet bin, Bluetooth-Untersützung zu programmieren. Wenn es Dir recht ist, melde ich mich in 2-3 Wochen...

Grüße,

Stefan.

kebekus, 2024-05-18:
Lieber Richard,

durch Zufall bin ich auf ein Problem aufmerksam geworden, dass vielleicht erklärt, wieso es Probleme mit der Wi-Fi-Verbindung gibt: Manche Geräte senden die NMEA-Sentences nicht in einem Stück, sondern in mehreren Blöcken (die sich nicht einmal an den Grenzen der Sentences orientieren). Ich habe Enroute beigebracht, mit diesen Daten umzugehen.

Könntest Du bei Gelegenheit vielleicht einmal die aktuelle Beta-Version (=2.30.22) testen, die derzeit bei Google Play bereitsteht?

Grüße,

Stefan.

richj1000, 2024-05-18:
Hallo Stefan, Danke für den Hinweis. Ich werde die neue Version testen und mich dann melden. Grüße Richard Von: Stefan Kebekus ***@***.***>Gesendet: Samstag, Mai 18, 2024 9:43:13 AMAn: Akaflieg-Freiburg/enroute ***@***.***>Cc: richj1000 ***@***.***>; Mention ***@***.***>Betreff: Re: [Akaflieg-Freiburg/enroute] Add IP address of "PowerFlarm Eagle" (lxnavigation) 192.168.4.1 to the list of traffic data providers or make the IP / Port user configurable (Issue #404)
Lieber Richard,
durch Zufall bin ich auf ein Problem aufmerksam geworden, dass vielleicht erklärt, wieso es Probleme mit der Wi-Fi-Verbindung gibt: Manche Geräte senden die NMEA-Sentences nicht in einem Stück, sondern in mehreren Blöcken (die sich nicht einmal an den Grenzen der Sentences orientieren). Ich habe Enroute beigebracht, mit diesen Daten umzugehen.
Könntest Du bei Gelegenheit vielleicht einmal die aktuelle Beta-Version (=2.30.22) testen, die derzeit bei Google Play bereitsteht?
Grüße,
Stefan.

kebekus, 2024-06-02:
Version 2.31.01, which is available at Google Play for beta testers, supports configurable IP addresses. Please test.

richj1000, 2024-06-07:
Hallo Stefan,

es hat etwas gedauert aber nun bin ich zum Testen gekommen.

Eine Verbindung per WLAN lies sich nicht herstellen. (IP 192.168.4.1 UDP
Port 2000). Die Ursache für das Problem mit WLAN konnte ich nicht
herausfinden - die Meldung war, dass der Verbindungsaufbau zu lange dauert.

Eine Verbindung per Bluetooth hat auf Anhieb funktioniert - super

Ich hatte ein ähnliches Verhalten mit VFR Nav - WLAN hat nicht stabil
funktioniert aber Bluetooth läuft sehr zuverlässig.

Auf jeden Fall ein großes Dankeschön an Dich und Deine Mitstreiter für
das Bereitstellen der Software und die schnelle Rückmeldung auf Anregungen!

Grüße

Richard

kebekus, 2024-06-08:
@richj1000 Danke für die Rückmeldung. Ich schließe den Vorgang jetzt. Bitte melde Dich bei weiteren Problemen jederzeit gern!

closed 2024-06-08 by kebekus