Akaflieg-Freiburg/enroute #408: calculate Density Altitude

charlylima, 2024-04-13:
I would like to suggest that the app calculates and shows density altitude next to the Weather data. 

**Is your feature request related to a problem? Please describe.**
As a Pilot you should calculate Density Altitude before flight to decide if the runway length is sufficient for a safe takeoff. 
It might be that in the morning when it is still cold from the night, you can easily start from a runway in the alps, later on a sunny day with high temperate it is getting critical because the engine is loosing power with increased temperature and lower density altitude. It is often surprising for pilots when you got used to the weaker engine power in summer and then start on a cold day how much better it performs. So to do a good flight preparation you would have to calculate density altitude and estimate the takeoff distance using the tables in the aircrafts manual. But in reality, calculating Density Altitude manually is too much effort, Hobby Pilots often dont do it. 

**Describe the solution you'd like**
The App should calculate Density Altitude, based on Elevation, QNH and Temperature. 
It could show it below METAR data. 

**Describe alternatives you've considered**
Alternatives:
* Manually extract weather data from Enroute and input it into a web based density altitude calculator
* Write an extra app for flight preparation. It would also need to download METAR data and have the elevation data of the airfields. 
* Add the functionality to calc Density Altitude to Metaf Library and then add it to Enroute

**Additional context**
A good explanation of the formulas is on this page:
https://aerotoolbox.com/density-altitude/

It could look like this:
[image]

What do you think? 
And where would you add it in the UI?

My branch, first try:
https://github.com/charlylima/enroute/tree/feature/densityAltitude

kebekus, 2024-04-14:
@charlylima Thanks! I will definitely come back to this. It will take me a few days or weeks, though, because I am currently implementing Bluetooth support for traffic data connections.

charlylima, 2024-04-20:
In case you like it, I would would then try to implement the more exact formula from aerotoolbox.com, currently on my branch is the simple linear formula that pilots use when manually calculating.

Also I would like to display a second value, the difference to the elevation of the airfield, like this: 
`  calculated Density Altitude: 2000ft (+500ft)`
Then the pilot could apply the value to nearby airports that do not have a Metar weather report themself.

It would also be interesting to estimate a factor by which the  takeoff distance will extend, but i am not sure if it is a good idea since it would be different for turbo-charged aircraft. There is the "Koch Chart" available that tries to calculate this factor.
(see [FAA document about Density Altitude](https://www.faasafety.gov/files/gslac/library/documents/2011/Aug/56396/FAA%20P-8740-02%20DensityAltitude[hi-res]%20branded.pdf))
(see this [interactive koch chart](https://www.takeofflanding.com/))

Also it would be possible that the Metar already contains the Density Altitude value, although it have never seen that in germany, we should detect that and then not display the calculated value. 

And I was considering to add the functionality to metaf library, but it might be not possible because the parameters of the function are like this:
`    calculateDensityAltitude(altFeet, qnh, temp_c);`
and I think the elevation of the airfield would not be available if you only have the METAR report as input.

kebekus, 2024-04-20:
@charlylima, I absolutely appreciate your input! Please go ahead with my blessings.

charlylima, 2024-09-27:
I did not have much time to work on it during summer (since we bought an experimental aircraft !!!), 
but i will continue to work on it during winter time.

kebekus, 2024-10-06:
@charlylima Roger. All the best wishes for flight with your new aircraft. Fly safe!

Munzijoy, 2024-11-04:
> Also I would like to display a second value, the difference to the elevation of the airfield
> calculated Density Altitude: 2000ft (+500ft)

I tried that format and especially for negative density altitudes and if the difference is negative, this could be hard to read.
> There is the "Koch Chart" available that tries to calculate this factor.

Here a rough approximation for the formula, used in the Koch Chart, is described. 
https://groups.google.com/g/rec.aviation.piloting/c/SDlMioBVqaA/m/8_x9GYsnGwAJ

I think the formulas are only valid for fixed pitched and constant speed propellors. I compared the results of the formulas to the results of the [graphical tool ](https://www.takeofflanding.com/) and they are similar at least.

Here's a suggestion for a layout, displaying those calculated values. For better readability maybe a bullet point list under the decoded METAR section would be useful. Here's a suggestion:
[image]

kebekus, 2024-11-05:
@Munzijoy I appreciate your suggestion! However, please give me a few days to better integrate your existing code into the app before coming back to your suggestion.

To keep separate things separate, could I ask you to open a new issue for your suggestion concerning Koch curves? -- Thanks!

Munzijoy, 2024-11-05:
I opened a new issue #463 for the calculation based on Koch chart, referencing the comment https://github.com/Akaflieg-Freiburg/enroute/issues/408#issuecomment-2455290535.

kebekus, 2024-11-07:
@Munzijoy @charlylima The feature is now available in the 'develop' branch.

kebekus, 2024-11-07:
@Munzijoy I could use help with the class Decoder. A static code check with Clang-Tidy shows many bugprone unchecked accesses to optional values. I lack the time to look at all that…

Munzijoy, 2024-11-07:
I will take a look at the Decoder class soon.

kebekus, 2024-11-07:
Thanks! Since this is now a cleanup task, the development for this will happen in the branch feature/cleanup.

kebekus, 2024-11-11:
Implemented in 2.32.0, which appeared on Google Play a few minutes ago. 

@schneisi will likely produce an iOS build within the next few days or weeks.

@charlylima @Munzijoy Thank you so much!

closed 2024-11-11 by kebekus