Akaflieg-Freiburg/enroute #410: Support XCVario NMEA Sentences

kebekus, 2024-04-19:
Title says it all…

charlylima, 2024-04-20:
I did not know XCVario and looked it up:
That is a great open source project implementing an electronic variometer with a display for gliders built with an ESP32 chip and they also sell the hardware built into a 61mm or 80mm case. 
It has sensors for energy compensated variometer, airspeed based on dynamic pressure, altitude based on static pressure, outside temperature, battery voltage, gyro sensors AHRS for accelleration and rotation. 
It interfaces via bluetooth, wlan, RS232 TTL serial port, CAN bus.
https://github.com/iltis42/XCVario
https://xcvario.com

Here is the documentation of the protocol:
https://xcvario.com/docs/nmea-protcols/#4-toc-title
https://github.com/iltis42/XCVario/tree/master/handbook

### XCVario protocol
```
Sentence has following format:
$PXCV,
BBB.B, // Vario, -30 to +30 m/s, negative sign for sink
C.C, // MacCready 0 to 10 m/s
EE, // Bugs degradation, 0 = clean to 30 %
F.FF, // Ballast 1.00 to 1.60
G, // 1 in climb, 0 in cruise, Note: Original Borgelt docu shows vice versa
HH.H, // Outside airtemp in degrees celcius ( may have leading negative sign )
QQQQ.Q, // QNH e.g. 1013.2
PPPP.P, // Static pressure in hPa
QQQQ.Q, // Dynamic pressure in Pa
RRR.R, // Roll angle
III.I, // Pitch angle
X.XX, // Acceleration in X-Axis
Y.YY, // Acceleration in Y-Axis
Z.ZZ, // Acceleration in Z-Axis
*CHK = standard NMEA checksum
<CR><LF>
```

It also supports these protocols:

### OpenVario
```
$POV,P,<baro>,Q,<dp>,E,<te>,T,<temp>“

Example: 
P,1018.35:  1018.35 hPa barometric pressure
E,2.3 :            +2.3 in m/s vario climb
T,23.52:        23.52° OAT temperature in deg C
```

### Borgelt
```
$PBB50,AAA,BBB.B,C.C,DDDDD,EE,F.FF,G,HH*CS<cr><lf> 
AAA = TAS 0 to 150 knots 
BBB.B = Vario, -10 to +15 knots, negative sign for sink 
C.C = MacCready 0 to 8.0 knots 
DDDDD = IAS squared 0 to 22500 
EE = bugs degradation, 0 = clean to 30 % 
F.FF = Ballast 1.00 to 1.60 
G = 1 in climb, 0 in cruise, Note: Original Borgelt docu shows vice versa 
HH = Outside airtemp in degrees celcius (may have leading negative sign)
CS = standard  NMEA checksum 

$PTAS1,xxx,yyy,zzzzz,aaa*CS<CR><LF> 
xxx: CV or current vario. =vario*10+200 range 0-400(display +/-20.0 knots) 
yyy: AV or average vario. =vario*10+200 range 0-400(display +/-20.0 knots) 
zzzzz: Barometric altitude in feet +2000, related to QNH standard 1013.25 setting 
aaa: TAS knots 0-200 
CS: XOR Checksumme
```

### Cambridge
```
!W,<1>,<2>,<3>,<4>,<5>,<6>,<7>,<8>,<9>,<10>,<11>,<12>,<13>*CS<CR><LF>

<1> Vector wind direction in degrees
<2> Vector wind speed in 10ths of meters per second
<3> Vector wind age in seconds
<4> Component wind in 10ths of m/s + 500 (500 = 0, 495 = 0.5 m/s tailwind)
<5> True altitude in Meters + 1000
<6> Instrument QNH setting
<7> True airspeed in 100ths of Meters per second
<8> Variometer reading in 10ths of knots + 200
<9> Averager reading in 10ths of knots + 200
<10> Relative variometer reading in 10ths of knots + 200
<11> Instrument MacCready setting in 10ths of knots
<12> Instrument Ballast setting in percent of capacity
<13> Instrument Bug setting
*CS Checksum, XOR of all bytes of the sentence after the ‘!’ and before the ‘*’
```

kebekus, 2025-03-04:
Implemented in 2.32.14, which appeared for beta-testing today. At present, the data is not used though.

closed 2025-03-04 by kebekus