Akaflieg-Freiburg/enroute #411: Support traffic data receivers connected via serial ports

kebekus, 2024-04-19:


kebekus, 2024-06-02:
Version 2.31.01 supports the feature. Please test!

kebekus, 2024-06-08:
It turns out that QtSerialPort does not work under Android -- it sees the Unix device files but cannot access them for lack of permissions. We might need to look at XCSoar to see what they are doing....

sfalmo, 2025-05-17:
This feature could be interesting for the SteFly Nav, which runs Android and has RS232 serial ports for connecting external location/traffic data receivers (it has no built-in GPS). Unfortunately, the option for serial connections is disabled by default in Enroute for Android. In our case, we would like to receive data from a Flarm device connected to one of the serial ports (`ttyS3`) via a RJ45 cable. This setup works in XCSoar, so it seems that the serial ports are configured to be accessible for applications without root permissions. If I understand the discussion in https://github.com/XCSoar/XCSoar/issues/1002 correctly, XCSoar does nothing special when connecting to `/dev/ttyS*`, it simply attempts to read from the device file.

QtSerialPort could possibly also access `/dev/ttyS*` on the SteFly Nav. I would like to test this, but could not find version 2.31.01 in the Releases and was not successful in compiling Enroute myself. If it turns out that this actually works, it might be useful to reenable this option on the Android build. Although it is admittedly a niche case to have RS232 serial ports on an Android device.

kebekus, 2026-01-02:
@hjr @sfalmo Frits van Breemen-Schneider kindly sent me a "XCTOUCHNAV" version XCTN7v3 as a test device for the development. The current beta-version of Enroute (= 3.0.52) is able to connect to serial port devices via USB and RJ45 ports, as long as the system permission allow access to the relevant device files. With that, I am now able to read data from an external GPS connected to one of the RJ45 ports.

If you have appropriate hardware, I would appreciate any reports. If you have not done that yet, follow one of the links below to enroll as a beta-tester. You can of course always leave the beta-test-program at any time.

- https://play.google.com/store/apps/details?id=de.akaflieg_freiburg.enroute
- https://play.google.com/apps/testing/de.akaflieg_freiburg.enroute

Looking forward to your feedback! -- Stefan.

sfalmo, 2026-01-13:
Sorry for the late reply. I can test this feature on the weekend and will report back.

kebekus, 2026-01-18:
Dear all, I believe the feature is implemented in Version 3.1.0, which appeared in production on Google Play a few minutes ago. For lack of a testing device, I was unable to try the feature on an XCNav device, but I can confirm that it works on the "XC Touch Nav".

@sfalmo Thank you for your help.

Stefan.

closed 2026-01-18 by kebekus

kebekus, 2026-01-18:
@sfalmo Roger. Und in welchen Flieger habt ihr das SteFly eingebaut? Und was sind die Erfahrungen? Ich bin echt neugierig, denn vielleicht sollte unser Verein das auch mal machen…