Akaflieg-Freiburg/enroute #412: Scroll/Zoom map with fingers does not work

MatBarley, 2024-05-15:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...' map
2. Click on '....'screen with one finger to move map, 2 fingers to zoom
3. Scroll down to '....' -
4. See error: map does not scroll/zoom (zoom with +/-button works well, moving map with flight progress works well), just with fingers does not work

**Expected behavior**
A clear and con: move/zoom map is expected. Shut down and open app again heals problem!
If applicable, add screenshots to help explain you to

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6] moto g84 5G
 - OS: [e.g. iOS8.1] Android 14
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.:
Happens sometimes, not always reproducible, around 4 times during 2:40h flight. If app got a Log/Debug-function please advise how to download log-file. Thanks

kebekus, 2024-05-17:
@MatBarley Thank you for the report! I will look into this, but I will need some time. Please be patient, I will get back to you.

tue-kyndal, 2024-06-26:
I  also see this or a similar behavior.  And its a magor problem for me. The app will now crash so much during flight its becoming an issue.

I often stall or crash the app, or lose the option to move and zoom the map. Its mostly if i twist the map during zoom.
Only a restart of the app will restore map move and zoom. 

Fix right now is to only move with one finger and zoom on + and -

I would prefer the 2 finger map twist option was or could be disabled, at least in North Up mode. It makes no sense to twist the map away from north. It just add confusion and make everything harder trying to fix the orientation etc or restart the app while flying.

Please, Just bind the map orientation to north up or track. The twist makes no sense.

NB: I have loaded a raster map. At first I thought it was the raster map either being badly rendered or to large. 
So I have struggled a lot trying to remake my map...But i think its unrelated to the raster map.

xxDUxx, 2024-07-12:
Sounds a lot like [#389](https://github.com/Akaflieg-Freiburg/enroute/issues/389)

xxDUxx, 2024-07-24:
Try disabling gesture navigation in Android settings. This does not appear to play well with Enroute Navigation.

TTC70, 2024-09-11:
Hello,
I do not know if the issue has been solved.
I can surely say that 1 finger scroll and 2 fingers zoom/move/rotate work perfectly on my Samsung Galaxy S10 (tried both on ground and in flight).

kebekus, 2024-09-16:
Another user reported the problem, and confirmed that disabling gesture navigation in Android settings solves the issue for him.

kebekus, 2026-10-18:
Thank you, @MatBarley, for reporting this, and thanks to everyone who added observations and workarounds here! I believe the problem with scrolling and zooming the map has been solved in the meantime, so I am closing this issue now.

Reports from real flights are extremely valuable to us, so please keep reporting whatever you notice. If the map stops responding to your fingers again, please re-open this issue and let us know which device, Android version and Enroute version you are using.

closed 2026-10-18 by kebekus