Akaflieg-Freiburg/enroute #413: tempo restr areas

Geitau, 2024-05-30:
.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.
Example is: ukraine conference 
the warningsign is shown, however its not possible to see the complete area affected by the notam.
Maybe by clicking on the valid period the affected area could be highlighted by a transparent circle.
That would help tremendously to reduce airspace violations.
Thanks for a great job you surely do

kebekus, 2024-05-30:
@Geitau Thank you for the report! The feature that you requested is on our TODO list, so I take the liberty to re-open your request. 

Big events, like the Ukraine conference, are currently covered because we update our airspace data daily, and the database at openAIP, which we use, typically reflects these changes in the airspace. For smaller events, this is however not always the case.

kebekus, 2024-12-25:
@Geitau We considered your suggestion and investigated possible implementations. Regretfully, it turns out that the data quality in the international NOTAM databases is so low that we cannot reliably extract airspace information from the NOTAMs without having a human looking at each and every single NOTAM. This is a service that we do not have the resources to provide.

Quite regretfully, we decided to close this issue and leave the situation as it is now.

No matter what: Best wishes. Merry Christmas!

closed 2024-12-25 by kebekus