Akaflieg-Freiburg/enroute #415: Switch over to fontawesome icons

kebekus, 2024-06-17:
One developer wrote: I noticed something interesting in the git directory of enroute:

From 6.5G, which is not small to be honest, 3.7G are coming just from material icons. I would recommend moving to FontAwesome free icons. You can download from their site the whole SVG package and it's a couple Mb instead of Gb.

[image]

yorickreum, 2025-07-29:
I frankely would use some lightweight open-source icon set, not the big commerical ones from Google or FontAwesome... 

Suggestions: 
[Feather Icons](https://feathericons.com/)
[Lucide](https://lucide.dev/)
[Heroicons:](https://heroicons.com/)

kebekus, 2025-07-30:
@yorickreum This has been on my TODO-List for a very long time. However, our small team presently lacks the ressources to implement this improvement. Any help is greatly welcome!