Akaflieg-Freiburg/enroute #418: OGN Flarm server - no augmentation and localhost 4353 is not persistent.

tue-kyndal, 2024-06-26:
When connected to the OGN viewer app on my phone, I can connect to the data connection on localhost pot 4353.

But I don't get any augmentation of any flights/planes on the screen. Either with Vector or Raster map as background.

Is it not the hole point to see the gliders on the screen?

Also the connection does not seem to be very persistent.
When playing with it, I had to enter the localhost pot 4353 on every app start.
Since this is for the OWN viewer, could this not be a build-in listening port?

Thanks

kebekus, 2024-07-03:
@tue-kyndal Thanks for your report. Please apologize for the slow answer; I am on a business trip. I have never seen or heard of the OGN viewer app. What is it, and where can I find it?

Best, Stefan.

tue-kyndal, 2024-07-12:
OGN viewer is an android app to spot gliders.
It also has a server other apps can connect to. 

An example would be see you navigation. 
This app can also work as an flarm device if you want. You just enter your flarm device data and you go live on ogn and flight radar. 

http://wiki.glidernet.org/

https://live.glidernet.org/#c=45.00000,5.00000&z=10&s=1

I was under the impression that it was surported. 
It would work just like SafeSky App, except for gliders and for free.

Would be perfect for tow planes without flarm. 

NB. I often tow close to the airport, and the destination is not calculated because I'm to close. 

That is not good.. I always need a distance to the airport... So I know if older gliders can glide back. 

Please concider to show destination info regardless of proximity.

tue-kyndal, 2024-07-12:
[image]
[image]
[image]
[image]
[image]

tue-kyndal, 2024-07-12:
Here a few pictures to illustrate. Pasted above. 

Here I show the virtual super cup oy-tue taking off from the train station... Was on the way home on a train when posting 🤣
And it only works when moving. 

All done with a device created on http://ddb.glidernet.org/
The ogn viewer and see you navigator.  Android Apps 

Its the setup I plan to use because the super cup does not have a flarm.
my mobile can act as one. Better than nothing!

Food for thought.. Perhaps you can implement the same. And it would be free. 

Safety first 😎

kebekus, 2024-07-16:
@tue-kyndal Thanks. I am confused because I cannot find the OGN viewer app anywhere. Can you help?

tue-kyndal, 2024-07-16:
OMG...its been taken down
https://github.com/Meisterschueler/ogn-viewer-android?tab=readme-ov-file
see issue #39
I hope they resolve it soon. That is a really good app for glider enthusiasts.

Should be possible to side load
https://ogn-viewer-flarm-radar.en.softonic.com/android
Let me know if you would like me to post a backup apk

Anyway. Yesterday  flu to a private airfield in an old super cup with no transponder. 
My eta was a bit  uncertain, due heavy winds and some build up etc. I had to navigate.
I turned on my see you app, and the ground crew could follow me just fine on flight radar, and was there to pick me up when I arrived. Rather handy. And if I got into trouble, I could have notified ATC about my pretense on flight radar 24.  didn't though, I was just using standard station reports for the VFR flight.

Tue

tue-kyndal, 2024-12-22:
http://wiki.glidernet.org/start

still think it would be pretty great if you could see the flarm trafic on the app.
Especially when towing gliders.

tue

kebekus, 2024-12-25:
@tue-kyndal Thanks again. We are a bit confused by your answer. Why don't you connect the app to your traffic data receiver, for reliable and much better-quality data?

kebekus, 2025-05-11:
@tue-kyndal It seems that all works fine with the brand-new app "CCAS VFR-Kollisionswarnsystem"

- https://play.google.com/store/apps/details?id=aero.ccas&%3Bhl=de

I am therefore closing this issue now. Please feel free to re-open if problems persist.

closed 2025-05-11 by kebekus