Akaflieg-Freiburg/enroute #421: Aerodrome not visible at certain zoom levels

ubx, 2024-07-15:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Navigate to aerodrome LSTB ([Bellechasse)
2. Zoom in to 6 km : aerodrome LSTB not visible
3. Zoom in to 3 km : aerodrome LSTB visible
4. Zoom out to 10 km: aerodrome LSTB visible

**Expected behavior**
The aerodrome should be visible ab all zoom levels

**Screenshots**
[image]
[image]
[image]

**Smartphone (please complete the following information):**
 - Device: Samsung Galaxy S7
 - OS: Android
 - Version 8.0

kebekus, 2024-07-16:
@ubx Thank you for the report. I will look into the issue. Still, we would like your understanding that we will take substantial time. We are a small team and program the app in our spare time. Development is slow, and we must prioritize strictly.

kebekus, 2024-10-06:
@ubx Update: I will start to work on this issue now.

kebekus, 2024-10-06:
@schneisi Will backport MapLibre style from external JSON into QML, which should give a better fine-tuning about drawing priorities, and as an additional benefit allows to change map properties dynamically. We had something like that in Qt5, and needed to change that for Qt6. Now MapLibre has caught up and we can revert to the old version.

kebekus, 2024-10-11:
The issue is fixed in version 2.31.12, which has become available at Google Play and will hopefully become available for iOS devices in the next few days or so. I am closing this issue now

@ubx Dear Andreas, thanks again for your report. Please feel free to re-open this issue if you are not satisfied with the fix. Please keep reporting issues that you become aware of.

closed 2024-10-11 by kebekus

ubx, 2024-10-13:
Perfect! Thanks for the fix.