Akaflieg-Freiburg/enroute #423: Issues with usability for parashooter drop flights

tue-kyndal, 2024-07-16:
I was just trained to be a parashooter drop flight pilot.

I was obviously using Enroute.

I had some  issues I what to share, primarily because of lacing information and the fact that Enroute for some strange reason stops calculating details when you are close to the destination.

When doing a drop, you fly downwind from the drop point then climb precisely into the wind towards the point, witch you must reach an a pre-set altitude.

You must calculate everything on the fly, using your current and ascend info (E.g 500'/min), time and distance to the point.

Procedure.
Before reaching the point you call out 3 min to drop to wake everybody up...ifs its 10 km drop its a long ascend!
Then 2 min to drop inc. a call to ATC for clearance.
1 min to drop..where everyone gets on their toes.
At zero..you throttle back, stabilize overhead and open the door.
Then you continue into the wind a precis number of miles...and the drop master looks out and confirms location and correct upwind distance etc.
Then everybody jumps ;O)

For this you need at all times,  even crossing overhead.

1: Distance to point
2: Time to point in minuses
3: The vector, and proper auto zoom to zero the map and zoom so you can see your precise point and location etc.

Its  absolutely vital that the calculations does not stop when you are in the vicinity of the drop point. 
The app must never stop giving you the info you need. 

I managed with the 5min internals on the vector and manual zoom, but it would must be super nice if the app actually did what I ask it when I set up a direct to point with rubber band calculations, and some intelligent auto-zoom when inbound a specific point.

NB:
It was absolutely super fun ;O)
https://www.facebook.com/share/p/a52JJFyETWfq3png/

kebekus, 2024-07-21:
@tue-kyndal As before: Thanks for your suggestion! We would like to support your use case ... but I should definitively point out that we will not be able to act quickly. We are a small team, and we are currently overwhelmed with user reports, changes in Apple/Google policies that we need to react to, a move of our complete infrastructure, etc etc

Without more people helping, I will definitively take substantial time before we can do anything here.

tue-kyndal, 2024-12-06:
Any chance this request will be looked at.

Its a major issue if you are dropping parachutes preciously over a spot, or towing gliders ...precisely inside a predefined aerobatic box, where you need to release them at preciously 5000' AGL on the border of a 1x1km box.

The real time calculation and timing include your current assent rate, distance to target in minutes, and then you adjust your inbound track to make the altitude bang on when you hit the target. 
But when the display of values is removed within 3 Nm you are lost, and have to just eyeball it on the vector. Come on. How hard can this be.

Could the "near destination" black out of values, not be a feature you could set in global settings, and then check some mark, if you need super near calculations for precise drops / aero-tows etc.
Thanks for a great software.
TK