Akaflieg-Freiburg/enroute #424: Allow to use syncthing etc for sharing aircraft, routes, waypoints between devices

baloan, 2024-07-26:
**Is your feature request related to a problem? Please describe.**
I cannot find / do not have access to the location where EnrouteVFR save sits internal data. It may be in Android/data/com.akaflieg-freiburg.enroute but I don't see any files in there.

**Describe the solution you'd like**
Please make the data directory for aircraft, routes, waypoints configurable so I can point my file synchonisation solution (syncthing or alike) to that directory.

**Describe alternatives you've considered**
I failed to save the routes by using Share. My Andriod didn't offer to save the GPX or GeoJSON to a directory.

kebekus, 2024-07-27:
@baloan Lieber Andreas, die Nachricht ist angekommen, besten Dank! Allerdings sind wir wir nur ein sehr kleines Team. Solange nicht mehr Leute mitmachen, ist die Entwicklung langsam und wir müssen streng priorisieren. Es wird deshalb sicher noch erhebliche Zeit brauchen, bis die Funktionalität erscheint.

Vielleicht hast Du Interesse, mitzuprogrammieren?

Grüße,

Stefan.

kebekus, 2024-07-28:
@baloan Anmerkung: Wie du richtig feststellst, sind die Verzeichnisse mit den Daten von Enroute derzeit privat. Wir hatten schon einmal drüber nachgedacht, die Verzeichnisse konfigurierbar zu machen, allerdings sind die Regeln von Google für Zugriffe aus externen Speicher kompliziert. Jede Hilfe ist willkommen