Akaflieg-Freiburg/enroute #427: transparent setting for bearingless airplanes

OdomLinx, 2024-08-05:
**Is your feature request related to a problem? Please describe.**
In some airspaces, e.g. RMZ, the map displays are sometimes difficult to recognize for some people in bearingless aircraft

**Describe the solution you'd like**
in file src/qml/items/MFM.qml line 296 change from:
 opacity: 0.3
to:
 opacity: 0.1

**Describe alternatives you've considered**
Possibly also adjustable via the menu

**Additional context**
here screenshots for comparison
bearingless traffic opacity =0.3
[image]

bearingless traffic opacity =0.1
[image]

bearingless traffic with alert opacity =0.3
[image]

bearingless traffic with alert opacity =0.1
[image]

kebekus, 2024-08-06:
Thank you for the suggestion. We will definitively consider that, but we will need a few days. At the moment our ressources are used 100% by enlarging the infrastructure, to make NOTAMs work reliably again. We will come back to you as soon as we can find the time.

kebekus, 2024-08-06:
@OdomLinx Do you want to join the development? We are definitively looking for help!

OdomLinx, 2024-08-06:
> @OdomLinx Do you want to join the development? We are definitively looking for help!

I'm not the best programmer, I never really got beyond "C++ in 21 days". It's enough to customize a few lines.

kebekus, 2024-08-09:
Implemented. The fix will be contained in the next release.

@OdomLinx Thank you for your help! Please keep reporting issues and suggestions for enhancement. Feel free to submit pull requests.

closed 2024-08-09 by kebekus