Akaflieg-Freiburg/enroute #431: No VACs visible on map since App update

Maurer-1, 2024-08-16:
Since I have updated the App on IOS to version 2.31.8 first all of my formerly saved VACs disappeared, the library was empty.
So I started to re-import my existing VACs (as geo-tiffs) and they appear again in the library, but when I choose them in the menu none of them appears on the map.
Can you please check and if possible fix that issue?

Thanks
Stefan

kebekus, 2024-08-16:
@Maurer-1 Ouch! This should not happen!

@schneisi Simon, kannst Du bitte einmal schauen, was da los ist? Kannst Du den Fehler reproduzieren?

kebekus, 2024-08-16:
@Maurer-1 Thank you for the report. We will look at the issue ASAP.

schneisi, 2024-08-16:
I can reproduce the issue

schneisi, 2024-08-16:
The appearance on the map will be fixed with the next release.

I could also reproduce the empty library after an app update, but it takes a little longer to dig into this.

Maurer-1, 2024-08-16:
Great.. thx for the quick problem solution. It is not so much effort to import the VACs again, as long as they appear on the map again.

thanks again

stefan

Maurer-1, 2024-08-19:
With the new IOS App Version 2.31.9 the visibility of the VACs on the map works again.
Great Job... thanks !!!!

kebekus, 2024-08-20:
@Maurer-1 Good to hear that the problem has been resolved. I am closing this issue now. Feel free to open more issue reports if you find any problems!

@schneisi Thank you for the quick reaction.

closed 2024-08-20 by kebekus