Akaflieg-Freiburg/enroute #439: Flight plan info on magenta line

TTC70, 2024-09-11:
A clear and concise description of what the problem is.
I would limit the distraction while flying adding a little more information on display regarding the specific leg of the flight plan.

**Describe the solution you'd like**
I would like to have a small box embedded on the single leg where I can read distance, magnetic heading and altitude (data to integrate, in case, in the flight plan).
In order not to clog the display, this box could be visible only, e.g., with zoom less than 5nm.
The photo attached is of a 5inch display of my Samsung S10.

[image]

**Describe alternatives you've considered**
At the moment the only alternative is to switch the display of the device to the text flight plan, which is a relatively big distraction for the pilot, especially while flying alone and without A/P.

kebekus, 2024-12-01:
@TTC70 Thanks for your suggestion. We thought about this and experimented a bit. It seems that a box that is big enough to read will typically clutter the display quite a bit. We are, however, very positive about your alternative suggestion to make the 'Remaining Route Info' expandable (https://github.com/Akaflieg-Freiburg/enroute/issues/455), and will probably implement this in one of the next releases.

closed 2024-12-01 by kebekus

TTC70, 2024-12-06:
Hey Stefan,
I understand the graphic Issue.
As mindset as a pilot, I like to be able to choose what I would like to see on the map while flying.
I still believe that showing those information on the moving map could be a game changer for a pilot flying alone.

I am not trying to get features like other apps (I am and I will be faithful to Enroute, for me is fantastic), and I believe Enroute has a huge margin for getting better, as ULM pilot I realized that in such small airplanes there is not much room for "tools", so having these info more or less immediately available is a less burden for the pilot.

To get back to Enroute, an additional way for showing these info on the moving map is to include the option for the pilot to see such info or not.
The choices would be in Settings -> Moving Map Section -> Below "Sailing sectors", and could be:
- show navigation box info (time - heading) on moving map Y/N;
- show time Y/N;
- show heading Y/N;
- below what zoom -> slide from 0 to 30km (or 0 to 30 miles according to the app settings).

Adding, maybe, also the possibility to choose the transparency of the info box (in the screenshot following here the transparency is set at 50%).

Following here a conceptual screenshot.
[image]