Akaflieg-Freiburg/enroute #440: Map intially not visible on phone disconnected from any network

kebekus, 2024-09-16:
One user reported:

> Ich habe auf ein neues Handy gewechselt, das Android 13 besitzt, da mein altes (Android 5) nicht mehr mit der Enroute Flight Navigation updaten wollte.
> 
> Nun beobachte ich, das ich die App im Flugmodus zwar starten kann, aber es wird keine Karte eingeblendet. Es bleibt weiß, der blaue Punkt in der Mitte ist zu sehen, aber weder Karte noch Airspace-Informationen sind zu sehen. Erst wenn ich den Flugmodus ausschalte und ein WLAN empfangbar ist, funktioniert es.
> 
>  Das ist deshalb blöd, da ich dieses Handy ausschließlich als Navi für das Fliegen nutze und es daher keine SIM Karte hat.
Wenn ich nun also im Flieger auf dem Vorfeld das Handy einschalte und dort kein WLAN empfangbar ist, kann ich die App nicht nutzen. Oder: Wie neulich passiert: ein Neustart des Handys in der Luft wegen Überhitzung nötig ist, müßte dort oben in der Luft irgendwie erst einmal Internetempfang hergestellt werden, bevor die App startet.

and continued:

> Kurz vorab: Mit Flugmodus meine ich das Flugzeugsymbol am Handy, also das Ausschalten von WLAN, Bluetooth, Handyempfang etc. GPS bleibt dabei angeschaltet. Ich nutze es um Akku zu sparen, da ich in diesem Handy keine SIM Karte eingelegt habe. Ich nutze es nur als Navi zum Fliegen.
>
> Aber der Fehler läßt sich auch anders reproduzieren:
>
> Ich stehe auf freiem Feld. Bester GPS Empfang. Weit und breit kein WLAN in das sich das Handy einloggen kann. Ich habe aber diesmal nichts abgeschaltet (Flugmodus) oder so, sondern bin einfach nur mit dem Handy in der Tasche von zu Hause weggefahren.
>
> Wenn ich nun Enroute Flight Navigation starte, wird die Karte und die Airspace-Karte nicht geladen. siehe Screenshot …….364.jpg Es scheinen aber Positionsdaten per GPS vorhanden und richtig zu sein. Zu sehen ist ein blauer Punkt. Auf Screenshot …362.jpg und …358.jpg ist dies zu sehen. Die GPS Koordinaten sehen korrekt zu meinem Standort aus. Außerdem kann ich auf Anflugkarten schalten, den benachbarten Flugplatz Schönhagen auswählen und dann wird er mit realistischer Position eingeblendet. Screenshot ….352.jpg
>
> Wenn ich nun mit meinem richtigen Handy einen WLAN-Hotspot mache und mein Navi-Handy verbinde, dann startet Enroute Flight Navigation korrekt. Screenshots 349, 345 und 342
>
> Es scheint also zum initialen Starten der App eine Datenverbindung nötig zu sein. Ist da richtig ?
>
> Vermutlich fällt dies bei den meisten Usern nicht auf, da sie ihr normales Alltagshandy verwenden, das sowieso per SIM Karte mit dem Internet verbunden ist.

kebekus, 2024-09-16:
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Gunnar32, 2024-09-21:
Ich konnte den Fehler jetzt beheben. 

Wegen man bei der Berechtigung der App nicht nur den Zugriff auf WLAN erlaubt, sondern zusätzlich auch den Zugriff auf mobile Daten, dann geht es. 
Obgleich ich bei diesem Handy überhaupt keine SIM Karte eingelegt habe und daher auch keine mobilen Daten bekomme, aber für die App scheint das wichtig zu sein. 

Jetzt läuft es!

kebekus, 2024-09-21:
@Gunnar32 Danke für die Meldung, das ist extrem hilfreich! Ich denke, das Problem ist folgendes: Enroute verwendet zwar keine Mobilen Daten, aber Enroute verwendet TCP/IP Netzwerkprotokolle, um innerhalb des Gerätes Daten zwischen den einzelnen Programmteilen auszutauschen. Das gilt insbesondere für die Daten, die in der Karte dargestellt werden. Bei Deinem Gerät könnte es sein, dass das TCP/IP-Netzwerk für Enroute einfach komplett abgeschalten oder gesperrt wird.

Ich denke, dass ich einen Test auf funktionierendes TCP/IP-Netzwerk einbauen sollte, damit die App den Nutzer warnen und ggfs Hilfestellung bei der Konfiguration anbieten kann. Das wird allerdings ein wenig Zeit in Anspruch nehmen,

kebekus, 2024-10-06:
@charlylima Understood. This might indeed be the problem.

If charlylime is correct, then a workaround will be difficult because we won't get the permission. I guess the best we can do is to check the connectivity and warn the user if things fail. It will take me a while to implement anything here because we have a long backlog of things to do.

kebekus, 2024-10-11:
Note to self: Qt 6.8 supports http requests via local sockets, which might be a workaround for the problem.

kebekus, 2026-10-21:
@Gunnar32 @charlylima Good news: this is fixed in **enroute 3.5.7**! The moving map no longer stays blank when the device has no network connection — for instance on phones without a SIM card that are used only for navigation, or when the system settings deny the app the use of mobile data. Enroute now works fully offline in these situations, as it always should have.

Thank you both so much for your patience and for the detailed reports — Gunnar's discovery of the mobile-data workaround and Charly's research into Android's restricted networking mode were key to tracking this down. It took a while, but your help made the fix possible.

**Deutsch:** Das Problem ist in **Enroute 3.5.7** behoben. Die Karte wird jetzt auch ohne Netzwerkverbindung angezeigt, also auch auf Handys ohne SIM-Karte oder wenn die App keine mobilen Daten nutzen darf. Vielen Dank für Eure Geduld und die hilfreichen Hinweise!

I'm closing this issue now. If the problem persists after updating, please don't hesitate to re-open it — we'll be glad to take another look. Happy and safe flying! ✈️

closed 2026-10-21 by kebekus