Akaflieg-Freiburg/enroute #441: IS INCORRECT? Map and VFR reporting point issue

benniAvia, 2024-09-17:
Hi all, 

So, not sure if the map issue is a bug issue, but somehow related to the app.

**Describe the** bug
Finland EFUT airport, mandatory VFR reporting PULJU, is marked in the AIP South East of the airport.

In Enroute in shows on the North West side of the airport, on the totally wrong side.

Out of interest, are these data downloaded from somewhere, should this be informed, as obviously this could have some impacts on the planning and the flying?

Attaching the official AIP map here, and the snapshot.

[AIP EFUT.pdf](https://github.com/user-attachments/files/17030183/AIP.EFUT.pdf)

[image]

kebekus, 2024-09-18:
@benniAvia Thanks for the report. It seems that you have discovered a bug in the database "open flightmaps" that Enroute uses to generate its maps. This is not an error in Enroute, and there is nothing I can do at this point. I will therefore close the issue here.

Could I please ask you to get in touch with people at "open flightmaps"? Open the app "Enroute", open the main menu and choose the entry "Bug report". Then, choose "Aviation Data". The page will then explain how to contact Peter Kemme, who is the right person to help.

closed 2024-09-18 by kebekus

benniAvia, 2024-09-18:
Thanks Stefan, I have now reported this to Peter. I hope fliers would check the official data as well, otherwise, this could lead to an error in planning!

 @kebekus

kebekus, 2024-09-18:
@benniAvia Sure. Have a look at the manual: https://akaflieg-freiburg.github.io/enrouteManual/01-gettingStarted/01-think.html#think-before-you-fly

> Always use official flight navigation data for flight preparation and navigate by officially authorized means. The use of non-certified navigation devices and software like Enroute Flight Navigation as a primary source of navigation may cause accidents leading to loss of lives.

This is true for any app.