Akaflieg-Freiburg/enroute #442: Add a way to change language in app

jakubkuba97, 2024-09-30:
There is currently no way to change the language in app, it always displays in the system language which can be annoying for multilingual people flying with various linguistic groups

The easiest solution should be to add the app to the list of languages shown in system, languages, apps

Occurs on android did not check on any other devices

kebekus, 2024-10-01:
@jakubkuba97 Thank you for your suggestion. We will consider that, but I can not promise that we can implement the feature soon. We are a tiny team and are currently overwhelmed by requests. Please be patient.

kebekus, 2024-10-13:
With the latest development version, users on Android 13+ can use the Android settings to set an app-specific language [previously, setting an app-specific language was not offered because Enroute did not support this feature]. Enroute detects the language change and asks the user to restart the app at a convenient time, for the language change to take effect.

kebekus, 2024-10-13:
@schneisi I understand that iOS also supports app-specific language settings. I this currently supported in any way? If not, how much effort will it be to support that? In essence, we need some platform-dependent code that detects language changes and calls an appropriate C++-Method in Enroute.

schneisi, 2024-10-13:
It's already fully supported in iOS. When the language changes, iOS automatically kills the App

kebekus, 2024-10-13:
@schneisi Roger. Thank you for the quick reply. Maybe I should follow that example and just kill the app.

kebekus, 2024-10-23:
@jakubkuba97 I released version 2.31.14 a few moments ago, where the app-specific language settings of Android 13+ are fully supported. Thank you for your suggestion! Please keep reporting any issues that you may find.

closed 2024-10-23 by kebekus