Akaflieg-Freiburg/enroute #443: Import Export of Filght Plans in different formats

TTC70, 2024-10-06:
**Is your feature request related to a problem? Please describe.**
It is not exactly a problem, but a feature which could ease the work of many pilots.
At the moment the only two versions of Flight Plans managed by Enroute Flight Navigation, are .GeoJSON and GPX.

**Describe the solution you'd like**
It could be very helpful to increase the compatibility of the app adding the feature to export or import a *.pln and *.fpl flight plans.
The following open source application already takes into the account the conversion between GPX and FPL format.

https://github.com/abdullah-radwan/PlansConverter

Could it be an inspiration for the programmers?

**Describe alternatives you've considered**
The only alternative is to use an "out of the app" application and convert the flight plan "off-site" (or "off-app").

kebekus, 2024-10-08:
There is excellent news. Heinz Blöchinger has kindly volunteered to help us implement this feature.

@bloech1 Thanks!

kebekus, 2024-10-08:
Development happens here: https://github.com/Akaflieg-Freiburg/flightRouteImport

kebekus, 2024-10-31:
Implemented in the development branch. Currently undergoing testing.

TTC70, 2024-11-01:
Stefan,
as far as today I could NOT load the Flight-plan.fpl into EFN. 
The error i get from Enroute is: Error in reading of file 'content://com.android.providers.downloads.documents/document/msf%3A17935' (I guess the path is related to my phone).
I created a simple flight plan on Skyvector, saved the flight plan into a .fpl file.
It is a simple flight plan:  LIRE - LIRF -  LIRU - LIRL - LIRA and the content of the .fpl file is the following: 

<?xml version="1.0" encoding="utf-8"?>
<flight-plan>
  <created>2024-11-01T14:09:49Z</created>
  <waypoint-table>
    <waypoint>
      <identifier>LIRE</identifier>
      <type>AIRPORT</type>
      <country-code>LI</country-code>
      <lat>41.659372</lat>
      <lon>12.445156</lon>
      <comment />
    </waypoint>
    <waypoint>
      <identifier>LIRF</identifier>
      <type>AIRPORT</type>
      <country-code>LI</country-code>
      <lat>41.800278</lat>
      <lon>12.238889</lon>
      <comment />
    </waypoint>
    <waypoint>
      <identifier>LIRU</identifier>
      <type>AIRPORT</type>
      <country-code>LI</country-code>
      <lat>41.951944</lat>
      <lon>12.500833</lon>
      <comment />
    </waypoint>
    <waypoint>
      <identifier>LIRL</identifier>
      <type>AIRPORT</type>
      <country-code>LI</country-code>
      <lat>41.546906</lat>
      <lon>12.908264</lon>
      <comment />
    </waypoint>
    <waypoint>
      <identifier>LIRA</identifier>
      <type>AIRPORT</type>
      <country-code>LI</country-code>
      <lat>41.799444</lat>
      <lon>12.597222</lon>
      <comment />
    </waypoint>
  </waypoint-table>
  <route>
    <route-name>LIRE LIRA</route-name>
    <flight-plan-index>1</flight-plan-index>
    <route-point>
      <waypoint-identifier>LIRE</waypoint-identifier>
      <waypoint-type>AIRPORT</waypoint-type>
      <waypoint-country-code>LI</waypoint-country-code>
    </route-point>
    <route-point>
      <waypoint-identifier>LIRF</waypoint-identifier>
      <waypoint-type>AIRPORT</waypoint-type>
      <waypoint-country-code>LI</waypoint-country-code>
    </route-point>
    <route-point>
      <waypoint-identifier>LIRU</waypoint-identifier>
      <waypoint-type>AIRPORT</waypoint-type>
      <waypoint-country-code>LI</waypoint-country-code>
    </route-point>
    <route-point>
      <waypoint-identifier>LIRL</waypoint-identifier>
      <waypoint-type>AIRPORT</waypoint-type>
      <waypoint-country-code>LI</waypoint-country-code>
    </route-point>
    <route-point>
      <waypoint-identifier>LIRA</waypoint-identifier>
      <waypoint-type>AIRPORT</waypoint-type>
      <waypoint-country-code>LI</waypoint-country-code>
    </route-point>
  </route>
</flight-plan>

bloech1, 2024-11-01:
Hello,can it be that the format of the file is wrong? The fpl format which is supported by Enroute is XML based. But the content you copied into your message isn’t. So, it looks like Skyvector doesn’t use the defined format even if it uses the extension fpl.Best regards, Heinz

  2024-11-01T14:09:49Z
  
    
      LIRE
      AIRPORT
      LI
      41.659372
      12.445156
      
    
    
      LIRF
      AIRPORT
      LI
      41.800278
      12.238889
      
    
    
      LIRU
      AIRPORT
      LI
      41.951944
      12.500833
      
    
    
      LIRL
      AIRPORT
      LI
      41.546906
      12.908264
      
    
    
      LIRA
      AIRPORT
      LI
      41.799444
      12.597222
      
    
  
  
    LIRE LIRA
    1
    
      LIRE
      AIRPORT
      LI
    
    
      LIRF
      AIRPORT
      LI
    
    
      LIRU
      AIRPORT
      LI
    
    
      LIRL
      AIRPORT
      LI
    
    
      LIRA
      AIRPORT
      LI

kebekus, 2024-11-01:
@TTC70 The feature is available in the development branch of GIT. If you want to test it, you will have to compile the programm yourself -- for now. I hope to have a version on Google Play by the end of next week.

TTC70, 2024-11-02:
Anyway maybe I don't have the version of Enroute which allows to load the PLN. Will wait for the Beta version.

kebekus, 2024-11-05:
Implemented in 2.31.16. Thanks for all your help!

closed 2024-11-05 by kebekus

kebekus, 2026-04-22:
Dear all, thanks for your comments and requests.

- Import of FPL and PLN **should work** can you please try this out in the latest version of Enroute and let me know if it worked for you? If it does not work, then this is a bug that I will fix.
- Export in FPL and PLN was never under consideration and is not implemented. We could consider supporting it if there is a convincing use case. Could you please tell us a little what you intend to use the feature for? Also, is there a use case not covered by the existing formats (GPX and GeoJSON)?

kebekus, 2026-09-15:
Version 3.5.0 is now available on Google Play, and the iOS version will follow shortly. Flight routes can now be exported in Garmin FPL and MSFS PLN formats. (I had announced this for 4.0.0 above; that release became 3.5.0.)

@TTC70 @adrianoellero @bloech1 @fbbln Thank you all for the suggestion and the discussion! Please try it with your avionics and simulators, and keep reporting anything you find.