Akaflieg-Freiburg/enroute #445: Display FL data in ft in addition to metres

DanD222, 2024-10-12:
At present, when the Vertical Distance is set to metres, the airspace borders on the map display the FL recalculated in metres - I think this is great.

I would also like to **see the original FL value in addition to the metre value**, for example 
“FL75 (2286 m)”

While taking up more space, it would 
- show the common / default value that’s displayed on ICAO maps
- allow for simpler radio communication between users with Vertical Distance set to metres and those using feet (e.g. “we need to pass under that FL75 airspace”)

[image]

kebekus, 2024-10-12:
@DanD222  Message arrived, thank you! We will consider your suggestion. Please don't expect a quick reaction, though. We are a small team. At present, we receive more requests than we can handle.

kebekus, 2024-10-13:
Transferring this issue to enrouteServer, because the implementation should take place in our infrastructure.