Akaflieg-Freiburg/enroute #446: Always show 4 characters of waypoint ID in top bar

DanD222, 2024-10-12:
**Describe the bug**
On an iPhone mini with a 5.4” screen, the destination waypoint is cropped to 2 characters, e.g. “LK..”

(This could be any airport in the country)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the main map
2. Click and hold on any airfield and choose it as the Route target
3. Observe the target ID in the top navigation bar

**Expected behavior**
I would like to see at least 4 characters to be always displayed

**Screenshots**
[image]

**Smartphone (please complete the following information):**
 - Devce: iPhone 13 mini
 - OS: iOS 17.7
 - Native app
 - Version 2.31.9

kebekus, 2024-10-12:
@DanD222  Message arrived, thank you! We will consider your suggestion. Please don't expect a quick reaction, though. We are a small team. At present, we receive more requests than we can handle.

TTC70, 2024-10-26:
If i may, it could be also useful if the user could choose the character dimensions.
Great app, btw

DanD222, 2024-10-28:
Having played with the app a bit more, I think it would be worthwhile to make space to always display (at least) 6 characters without truncating rather than 4, so that the names of Compulsory Reporting Points would be displayed in full as well.

kebekus, 2025-05-11:
@DanD222 The feature you requested has been implemented. Thank you for your help!

closed 2025-05-11 by kebekus

DanD222, 2025-05-11:
Excellent, thanks!