Akaflieg-Freiburg/enroute #447: WearOS

a533, 2024-10-20:
Good morning,

Congratulations on the excellent app. I use it in every flight. Have you considered adding watch Wear OS compatibility? It would be nice to be able to see some data in the watch, like a simple CDI, the GS, the heading , the FL or the ETE and distance to the next point.

Just a suggestion.

Thanks a lot 

Alvaro

kebekus, 2024-10-20:
@a533 Dear Alvaro, we would like to support WearOS. However, we are a very small team with limited ressources, and our calls for help have not led to any response from the community. Unless more people will join the team, it seems unlikely that we will be able to implement the feature anytime soon.

I will leave this issue open for a while, and close it if we can't find anyone willing to work on the issue. I am sorry that I have no better news.

a533, 2024-10-20:
Thanks

AQ

kebekus, 2025-05-11:
@a533 As I had feared, we have not found any volunteers to work on this, or donate the necessary hardware.

I am sorry that I have no better news and close this issue now.

closed 2025-05-11 by kebekus

a533, 2025-05-11:
I completely understand. Thanks anyway

Regards

AQ

iflyhere, 2026-09-14:
@kebekus — I have picked this up and would like to offer it back.

I have been building a Wear OS companion for Enroute over the past weeks, and I
have the hardware to test it on: a Galaxy Watch 7 and a Galaxy S24. The work is
in a public fork at https://github.com/iflyhere/enroute, branch
`feature/companion-protocol`, and it is structured so it can come back as a
series of reviewable PRs against `develop` rather than one large drop.

**The architecture point that matters most for this project: no Google Play
Services.** The obvious way to build a Wear companion is the Wearable Data
Layer, but that library ships under Google's proprietary Android SDK Terms, and
linking it into a GPL-3.0 app needs either a §7 additional permission — which
only the copyright holders can grant — or an argument that it is a "System
Library". It would also force the watch APK to carry the same applicationId and
be signed with the same key as the phone app. The whole question is avoided by
not using it.

The phone side therefore adds **no new dependency at all**: it uses
`Qt6::HttpServer` and `Qt6::Bluetooth`, both already linked. Everything travels
over one documented protocol (`doc/companion-protocol.md`), so the encoder knows
nothing about transports and the watch knows nothing about which one it is on.

### What works today

**Two transports.** HTTP/JSON on the local network, and BLE GATT with the phone
as peripheral — which is the one that matters in an aircraft, where there is no
Wi-Fi. The watch picks automatically or can be pinned to either.

**The data:** flight route, live navigation frame at 1 Hz, NOTAMs, weather,
approach charts, traffic, nearby waypoints, the flight log, and the pilot's
display preferences. Every displayed quantity travels twice — raw SI and the
phone's own formatted string — so the watch never disagrees with the phone about
rounding, units or translation.

**The moving map, from the phone.** Style, vector tiles, the aviation overlay in
the app's own colours, sprites and glyph ranges all come off the phone, so the
watch needs no internet connection and no map service of its own. This is also a
licensing property, not just a convenience: the openAIP and open flightmaps data
stays the pilot's own downloaded copy, and the watch's HTTP client refuses to
speak to any host but the paired phone.

**Approach charts** are drawn georeferenced on that map, selected the way the
phone selects one.

**Instrument faces** — altimeter, ASI, variometer — drawn to match real
instruments, plus a frequency page showing the radio frequencies of the route's
waypoints and the flight information sector the aircraft is currently inside,
which is the one a pilot cannot look up from a chart in flight.

**Off by default**, protected by a six-digit pairing code, with the listening
address and the client count shown in Settings.

Everything is configurable from the phone: page order, hidden pages, bezel
behaviour, link preference, alarm vibration, always-on display.

### Measured on hardware

- Bluetooth: MTU 517 negotiated, all nine documents in hand ~4.5 s after
  connecting, including a 152 kB approach chart index from a full AIP VFR
  Germany trip kit (519 charts) that takes 2.7 s of that.
- Map: returning to the map page from another screen costs no network request
  at all and redraws in about 1.2 s.
- A simulated flight (X-Plane UDP, the documented flight-simulator input) across
  a FIS sector boundary: waypoint advance, ETE/ETA countdown, chart entering and
  leaving range, and the sector frequency changing all behave.

Not measured: battery over a full flight. Not tested: iOS.

### Two small things found along the way

While reading the code I found two defects unrelated to this work, each on its
own branch in the fork and each a small standalone PR if you want them:

- `NOTAM::updateSectionTitle()` tests `addDays(90)` twice, so the `"> 90 days"`
  branch is unreachable.
- `NotificationManager::setupSpeaker()` reaches for a `GlobalObject` singleton
  from a `QtConcurrent` thread on Linux desktop, which trips the assertion in
  `allocateInternal` — 8 aborts out of 8 in a Debug build once a map is
  downloaded.

### How would you like it?

I know the team is small, and I would rather fit the way you want to work than
hand you something awkward to review. My plan was roughly: protocol and phone
server first, then the watch app, then the BLE transport, each standing on its
own. I am happy to split it differently, to keep the watch project out of this
repository entirely, or to maintain it in the fork if you would rather not take
on the maintenance — just say which.