Akaflieg-Freiburg/enroute #448: Satellite View - Option to use Google Earth instead of Google Maps

toficly, 2024-10-21:
**Is your feature request related to a problem? Please describe.**
On my Android Device I also use Skydemon which uses Google Earth to show the Satellite View of a Waypoint. I don't want to install Google Maps additionally for Enroute

**Describe the solution you'd like**
I would like to have a setting where I can choose where to open the Satellite View: Google Maps or Google Earth

**Describe alternatives you've considered**
I could also use a Web Browser instead of the App. But I would like to avoid installing a Web Browser as well

**Additional context**
Thank you for developing this wonderful software and making it available for free - very appreciated!

kebekus, 2024-10-21:
@toficly Interesting suggestion. I will consider Google Earth. However, I should point out that we are a small team only, so that development is slow. Please be patient.

kebekus, 2024-11-26:
Here's how this could be done:

- https://www.perplexity.ai/search/i-have-an-android-app-and-i-wo-XkGZnWhZTRaX25vj2LZZOw

kebekus, 2024-12-13:
@toficly Implemented in the 'develop' branch. A first version for beta-testers will likely appear on Google Play next week.

toficly, 2024-12-16:
@kebekus that was fast - thank you 👍 I'm happy to test

kebekus, 2024-12-23:
The suggestion is implemented in Version 2.32.3, which is now available for Beta-Testing on Google Play.

toficly, 2024-12-23:
On a Samsung Galaxy Tablet with only Google Earth installed running latest OS it opens directly without any further configuration and shows the correct image - tried different Airfields. @kebekus Well Done! From my side Ticket can be closed. Merry Christmas and a Happy New Year.

kebekus, 2024-12-24:
I am closing this issue because the improvement has appeared in the production version on Google Play today.

@toficly Thank you for the suggestion!

closed 2024-12-24 by kebekus