Akaflieg-Freiburg/enroute #449: Direct to on an active flight

lmarko76, 2024-10-21:
**Is your feature request related to a problem? Please describe.**
When flying along the route with a pre-programmed multiple waypoints if you receive a "Direct to" waypoint from ATC and use the long press to get the context menu and choose "Direct" the rest of the route will be deleted, leaving no route after the chosen direct waypoint.

More detailed explanation with a scenario:
A generic route looks like this: ADEP-WP1-WP2-WP3-WP4-WP5-ADES
Where:
- ADEP is departure airport
- ADES is destination airport
- WPx is a waypoint along the route (as filed in the flight plan)

Scenario:
The airplane departs ADEP and flies to WP1. When reaching WP1 it flies to WP2. Just before reaching WP2 the ATC comes on the radio and gives the instruction to fly direct to WP4. By long press on the WP4 and choosing the "Direct" the application draws the line direct to the chosen WP4 and deletes all the rest. So, the legs form WP4 to WP5 and destination are lost. But the ATC expect the airplane to fly the rest of the route as filed in the flight plan.
It would be nice to keep the route after the chosen WP4. It would be even nicer if the original route would remain on the screen but with a very thin line.

**Describe the solution you'd like**
The solution would be: that at the moment the user chooses direct to a waypoint to plot a route form the present position(PPOS) direct to the chosen waypoint but keeping the route after the "Direct to" waypoint intact. In the above scenario the route(after choosing "Direct" in the context menu) would be PPOS-WP4-WP5-ADES

**Describe alternatives you've considered**
An alternative would be that there is a new option in the pop-up(long press) menu that would keep the plan after the "Direct to" intact.
A similar behaviour described above could be achieved as when getting an direct form ATC the user opens the "Route and Wind" and manually deletes the points up to the direct waypoint. In my opinion it would be a bit to complicated especially during the flight.

**Additional context**
Behaviour on Androide device.
Other than that the app is perfect.

kebekus, 2024-10-21:
@lmarko76 Thanks for your suggestion. The feature has been requested numerous times lately, and is therefore rather high on our priority list. Still, I have to point out that we are a small team only, and therefore need time. Please be patient!

TTC70, 2024-10-26:
I don't think that this is an issue. 
When ATC asks you to go directly to WP4, Instead to press "direct to" on WP4, choose the next waypoint (WP3 in your case) and delete it so the flight plan will not be canceled but only slightly modified, so you have both magenta and "rubber line" pointing directly to WP4 without cancelling the whole plan.