Akaflieg-Freiburg/enroute #450: Add Caribbean Maps

papoanaya, 2024-10-21:
**Add Caribbean Maps** 

I am not sure why the Caribbean is not included as a region to be downloaded, but the data is available in OpenAIP. If there is a method that one can download the data and incorporate it on its own, then it would be nice to know if that options is available.

kebekus, 2024-10-23:
@papoanaya Dear Luis, the Caribbean is not included as a region because we live in Germany, have no clue about the airspace situation there, and no one interested in the area has contacted us yet.

If we can reasonably ensure decent data exists in openAIP, we will gladly support the Caribbean. Can you help? To begin, can you tell us a little about yourself? Where do you fly exactly, and what kind of aircraft do you fly?

kebekus, 2024-10-23:
@papoanaya I will transfer this issue to the repository "enrouteServer" because this is where our infrastructure code is stored. You might receive several notifications from GitHub in a few moments.