Akaflieg-Freiburg/enroute #454: Display FL from cabin pressure if no external static pressure available

DanD222, 2024-10-27:
**Is your feature request related to a problem? Please describe.**
At present, I see that my iPhone's barometric sensor is used to calculate and display the Cabin Altitude.
I have no external traffic receiver connected to the app:
[image]

The FL infobox on the map screen doesn't show a FL value:
[image]

**Describe the solution you'd like**
I would like to have the option to use the phone's internal barometric sensor data to calculate and display the FL value in the FL infobox on the map page.

**Additional context**
I'm not sure whether the current behaviour is by design - I can imagine it could be, in that it is desirable to calculate FL only from a "proper" static pressure from a static port. 

However, if the external traffic receiver is for example a Power Flarm, then I believe that the pressure reported by this device is also the Cabin Altitude, as the device does not have a static port connection and is essentially a box sitting in the cockpit, like the phone.

I could imagine the feature being selectable on the Barometric Data screen, plus also perhaps colouring the FL values in the infobox to indicate that the FL data is calculated from the phone's internal sensor and not from an external device.
This could help to make the user aware of the source the data and warn him to apply extra caution when maintaining vertical airspace separation.

PilotNubilus, 2025-05-09:
@DanD222 I had the same issue and could not solve it, but I found something, which is maybe enough for you. As I understand your problem you want to use the barometric sensor of your phone, because there is no external device providing static pressure. You could install the free App [CCAS VFR-Kollisionswarnsystem](https://play.google.com/store/apps/details?id=aero.ccas&hl=de_AT) on your phone. It is very simple. You just need to open it and as long as you have set a registration at some point (can be changed any time) it acts like a traffic receiver and sends traffic data and pressure data to Enroute, which are both displayed there. The traffic data is received from the internet via your phone's connection and the pressure data is collected from your phone's sensor. Your position and registration is shared via the internet as well, so that your are visible for others on their respective traffic displays. Most of the time it connects instantly, when I open Enroute. According to the description of the app ADS-B, FLARM and OGN data is shown as well.

I do not like the "iPad" pilots and always say look out look out look out. I already tried this app during flight with Enroute. On the one hand you could just ignore the traffic data, because it could be a distraction or not that useful, because it may lag behind or give you a false sense of saftey, but on the other hand you provide your position for the "iPad" pilots and you may actually find an aircraft that would have been missed otherwise. Not that long ago I was flying in an area that was a bit more rural than usual and it showed some traffic. After a few moments of looking at this specific position, I spotted a glider thermaling, that likely would have been missed otherwise. I can't give you a final verdict, but I had at least one positive experience, while using it.

best regards

DanD222, 2025-05-11:
@PilotNubilus Thanks for the tip! I wasn’t aware of the app - it works well as a workaround, both for iOS and Android.