Akaflieg-Freiburg/enroute #458: Support Check Lists

kebekus, 2024-11-01:
Investigate how check lists could be most easily supported.

TTC70, 2024-11-03:
I might have an idea.
Assumption
The checklists are proper and specific for owner's plane.
It is only a list of items/things to do/verify onboard.

Plan
My idea is to have the Voice "Checklists" in the main menu right after "Airplane" (see screenshot).
.
The checklists themselves could be created, set up and "filled" in the menù "Airplane" after all submenus (Name, Units, True airspeed, Fuel consumption, then Checklists).
The submenu "Checklist" could have another submenu where the user can choose how many checklists they want to create, how many items per checklist and finally write the items one by one.
For instance, i have the following checklists for my plane: Engine run-up, pre-taxi, Engine Check, Pre-alignment, take off, Engine off and, of course, emergency procedures.

Just an idea.

[image]

kebekus, 2024-11-03:
@TTC70 That's basically what I also have in mind. It will be a while, though, before I can find the time to implement anything here.