Akaflieg-Freiburg/enroute #460: can t connect

Perry17532, 2024-11-01:
[image]

MSFS 2020 is running. Cessna 152 is activ at Norderney (EDWY). fs2ll is running and connectet. port 2000 is open in firewall under "incomming rules" (Eingehende Regeln.). Error like screenshot is still happening. 

Are there a solution to fix it? Maybee i am wrong. Where?

Can you help, pls.

Have a nice day.

Perry

**Desktop (please complete the following information):**
 - OS: Win 11
 - Browser firefox
 - Version 132.0 (64 bit)

**Smartphone (please complete the following information):**
 - Device: Samsung S7 FE 5G(Tablett)
 - OS: Android 14
 - Browser standard on android 14
 - Version [e.g. 22]

TTC70, 2024-11-01:
You didn't write anything about FSUIPC7.
Do you have it running?

FS2FF uses FSUIPC7 in order to send airplane position and flight data to Enroute.

Perry17532, 2024-11-01:
don´t know. It is installed but if it is running? how to check?

TTC70, 2024-11-01:
> don´t know. It is installed but if it is running? how to check?
> 
> can´t find in Task-Manager: nore in Processe, not in Details or Services.

It should be in the tray bar. If you don't see it then you have to launch manually. FSUIPC7 is mandatory for FS2FF to work.
Anyway if you launch it and it is already in execution, the program itself will tell you that it is already running.

Perry17532, 2024-11-01:
i installed the newest fsuipc7 and started the sim with it. fs2ll is connected. Problem still exist.

TTC70, 2024-11-02:
> i installed the newest fsuipc7 and started the sim with it. fs2ll is connected. Problem still exist.

What i can tell you is that the connection is pretty straightforward.
Settings to verify:
- Device IP on the wifi network - both computer and device must be in the same mask 192.168.1.xx - for instance, computer 192.168.1.10 and device 192.168.1.15.
- MSFS, FSUIPC7 and fs2ff up and running.
I have this configuration and I don't have any issue at all.
Try to configure fs2ff manually, uncheck the "auto connect" box and write the IP of the device onto fs2ff manually under the field "DEVICE IP", uncheck the "Auto-detect" box.
The only reason the device is not connected to the PC (I am NOT talking about Enroute connected with MSFS/FSUIPC7) is because they do not have the same IP configuration.

Perry17532, 2024-11-02:
Hi there,
thanks a lot for your advice. I changed to a payware. This one works as soon as it was installed. I was too frustrated with the fs2ll. 
But anyhow. Thank you a lot.

By
Perry

closed 2024-11-02 by Perry17532