Akaflieg-Freiburg/enroute #463: Calculation of change in aircraft takeoff distance and rate of climb based on Koch Chart

Munzijoy, 2024-11-05:
**Is your feature request related to a problem? Please describe.**
Based on METAR information, an approximation of how how pressure altitude and temperature affect the increase of takeoff distance and decrease in rate of climb, could be automatically calculated. A so called Koch Chart can be used to estimate those values.

**Describe the solution you'd like**
Show the estimated increase of takeoff distance and decrease in rate of climb below METAR information.

For further information and a first suggestion on how this could look like, reference to this comment: _Originally posted by @Munzijoy in https://github.com/Akaflieg-Freiburg/enroute/issues/408#issuecomment-2455290535_

kebekus, 2024-11-05:
@Munzijoy Understood. As you probably saw, I am currently streamlining your code and refactoring the METAR class, which contails a lot of old code. I will come back to you as soon as I am finished.

kebekus, 2024-11-24:
Implemented in the 'develop' branch. Currently updating the manual accordingly.

kebekus, 2024-11-26:
Implemented in 2.32.2, which appeared on Google Play today.

@Munzijoy Thanks!

closed 2024-11-26 by kebekus