Akaflieg-Freiburg/enroute #468: Support geoclue d-bus interface for geolocation

ruzko, 2024-11-10:
**Is your feature request related to a problem? Please describe.**
I use Linux on the desktop, as well as on my phone (which I bring with me on flights).
Gpsd (which communicates with client applications using TCP/UDP, and therefore works with Enroute) used to be the standard geolocation provider on Linux.
Gpsd is deprecated for Gnome and KDE Plasma DEs, in favour of Geoclue-2. Geoclue communicates over D-bus, and allows for more granular control over which applications can access geolocation data, and what precision to allow them.
In some cases the GPS capability of a modem can be set to "unmanaged", allowing both gpsd and geoclue to work simultaneously, but this is not always the case.
My phone's modem doesn't support "unmanaged gps mode", and therefore doesn't allow me to run both gpsd and geoclue. I don't want to lose the boost to privacy and security that geoclue offers over gpsd. This leaves me unable to use Enroute.

**Describe the solution you'd like**
Support Geoclue-2 D-bus API as a geolocation source.

**Describe alternatives you've considered**
It's possible to implement a "bridge", which takes geolocation data from geoclue through D-bus and outputs it in a standard format over TCP or UDP. I've made a similar [bridge for waydroid](https://gitlab.com/papiris/geobridge) but it's more of a stopgap measure which doesn't address privacy or security concerns. 

**Additional context**
Lovely software, thanks for developing it! :)

My phone runs [postmarketOS v24.06](https://postmarketos.org), if it matters.

kebekus, 2024-11-11:
@ruzko Thank you for your suggestion. I understand your request, but I fear that we do not have the resources to work on this. At this point, I can only ask for your understanding. We are a very small team, so development is slow, and we must prioritize strictly. I regret that I do not have better news.

I plan to leave this issue open for a few days to see if someone (perhaps you?) is willing to work on this. If not, I will close it soon.

kebekus, 2024-11-17:
@ruzko As I had guessed, there isn't anyone willing to work on this. I am therefore closing the issue now. I regret that I do not have better news.

closed 2024-11-17 by kebekus